benb0jangles/FlySky-i6-Mod- #2: Cant find the Exe file for the i6 alt mod.. and i reallly meed it

Brun-X, 2016-01-01:


benb0jangles, 2016-01-01:
https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/Mod_i6_Alt_Recovery_Programmer_V1_0

Brun-X, 2016-01-01:
:( I DIDNT WORK

benb0jangles, 2016-01-01:
tried original firmware?

Brun-X, 2016-01-02:
yes the original firmware don't work either.. ( at least not the version i tried (thje only one i could find)     (V. 1.10)
Novax firmaware is the only firmware that works.. it totallly bricked....

every time i try to install some firmware it gives me the alarm for setting all switches to "zero"
pls help... im afraid i have to buy a new one. Even though this tx is brand new

benb0jangles, 2016-01-02:
never heard of anyone being stuck on "all switches to "zero"" before. Doesn't sound like a firmware issue, sorry & good luck.

Brun-X, 2016-01-02:
it seems like the flysky site is under construction and it wont enter the
site, so i can't download the original firmware from there.

but i found the original firmware some other place but i can't say for sure
if its the right one...

maybe you could send me a copy of the original firmware??

i haven't tried the CD that came in the bow woth the Tx cause i have no
working CD drive.. but i will try this when i get to a PC with CD drive..

thankful regards
Søren

benb0jangles, 2016-01-02:
All the Original firmwares are on the Github:
https://github.com/benb0jangles/FlySky-i6-Mod-

Brun-X, 2016-01-03:
the original tgy firmware from github doesn't work either..

i have no idea what to do next?

benb0jangles, 2016-01-04:
If you are new to quadcopters I don't recommend mods or firmware changing.

Brun-X, 2016-01-04:
i am new not entirely new to quads but new to fixed wings.. so novax
firmware is not desirable to flying fixed wing..

i did not know you where not supposed  to uddate the firmware..i thougt it
was like an iphone where you where ment to update it..

but anyway... it looks like i have to nuy a new one

benb0jangles, 2016-01-04:
not sure if trolling or serious

Brun-X, 2016-01-04:
im sorry if i havent made myself clear..

i am not trolling i just found github by accident and saw that you made the
alt mod firmware for i6 and hoped that you might have an answer to my
problem..

this is not bullshit i really have a problem

Brun-X, 2016-01-05:
Bullshit or not. Thank you for your time. Hope I can get an answer
elsewhere.
Happy new year
Søren Brun
Den 04/01/2016 19.34 skrev "Søren Brun" <sbrun8600@gmail.com>:

matrhint, 2016-01-20:
I believe the issue is that the program is not working with his computer. I am having the same issue. When I start the program as administrator I get a :NTVDM CPU Has Encountered an Illegal Instruction message for all of the programs for firmware flashing.

matrhint, 2016-01-20:
The issue is how it is downloaded. It seems that the program wouldn't properly download without being logged into github. Once I logged in i downloaded the same files and they had the icons for the files and worked properly once ran per the instructions.

benb0jangles, 2016-01-20:
Ahm, untrue, It does download & works when I downloaded using a different computer & not github registered. I just tested this theory.
-Windows 7, click link
-"This file has been truncated, but you can view the full file." - click the "view full file"
-File downloaded & works.
-simple as simple gets

Hope you solve your problems.

Brun-X, 2016-01-20:
hey thank you very much for you answer... i'm having a hard tine
understanding your instructions though :/

Brun-X, 2016-01-20:
im my case all the firmwares i downloaded all have their own logo (icon) so
i believe i have downloaded then  as you say

2016-01-20 8:13 GMT+01:00 Søren Brun <sbrun8600@gmail.com>:

benb0jangles, 2016-01-20:
try discussing with other users here: http://www.rcgroups.com/forums/showthread.php?t=2486545

Brun-X, 2016-01-20:
thank you

ive allready posted a thread on rcgroups and got a few replys.. but noon of
them solved the problem..

and now i ordered a new i6 wich i will not touch the firmware on :)
besides i found out how to get the maximum out of the novaX firmware.. i
would not say that i have given up but i cant think of any more to do about
it..

matrhint, 2016-01-20:
Oops. I believe you are correct. Thank you for your work. I am excited to try it all out today.

A-Nesh, 2016-04-09:
I have the same "all switches to zero" issue. I have band new FlySky FS-i6 and it worked fine until i upgraded it to "10ch_MOD_i6_Programmer_V1_5". Which totally bricked it. I have tried Flysky i6 Original Firmware, Turnigy i6 Original Firmware and all versions off 10ch Mod i6 Updater with no luck. I believe it is some kind calibration issue that throttle control doesn't read completely zero. Is it possible to disable "throttle to zero" upon startup, like in NovaX firmware?

appletaartje, 2016-08-28:
OH
MY 
GOD

Can it be so simple?? Why isn't there any manual on how to do this? People that don't know Github will probably be stuck for ages, because just clicking on the file (.exe) will only download the small version of the file (44KB) and not the "full" version (multiple MB's).

So, for me the simple solution is:

1. Leftclick on the "folder" in which you want to download a file (e.g. "10ch Mod i6 Updater").
2. Leftclick on the next folder (e.g. "10ch_MOD_i6_Programmer_V1_5")
3. Leftclick (not rightclick save as....!!) on the .exe file (e.g. "10ch_MOD_i6_Programmer_V1_5.exe")
4. A new page will be loaded and will show the real size of the file (e.g. 2.01 MB)
5. There is a BUTTON "Download". Left click on it......

Voila, file downloaded correctly and can be opened.
The other way around (the 44KB file) will not work, and gave me errors like: The version of this file is not compatible with the version of Windows you are running. And also like: NTVDM CPU Has Encountered an Illegal Instruction.

So this is what worked for me, nice to have found out myself, for nobody on this site could help.........

Cheers!

Brun-X, 2016-08-28:
well thank you for telling me.. i haven't had the same problem though.. i
just checked and all the firmware versions i have downloaded seems to be
around 2mb
my problem is that i tried all those versions and non of them works.. when
i try flashing any fw version i get the Tx equivalent of windows´ blue
screen of death.. that is the all to familiar beeping alarm like sound and
a screen that tells you that you need to place all sticks and switches in
their zero state.. like when you turn on your transmitter with throttle on..

everything i tried and i spend many many hours trying to make it work but i
haven't had any luck yet.. i guess i could give some of the new fw version
a shot but i'm afraid that it has to do with the sticks not being able to
get to their zero state maybe due to pre-programmed dual rates set too low
or maybe a faulty component.. but i just startet on an electronics
engineering education so i might have a shot learning more about it along
the way..

appletaartje, 2016-08-29:
Have you tried resetting the Tx back to factory defaults?
Can you even do that, or is that also not possible?

I think maybe its a malfunction of the hardware, instead of the software (original or modded)...

Brun-X, 2016-09-02:
well i bricked it when i flashed NovaX firmware and in that version there
is no option to reset to factory default :/

i'm pretty sure it has to do with hardware as well.. do you know if it is
possible to maybe short out the stick gimbals to ground (to zero them)
while i try flashing it again?