benb0jangles/FlySky-i6-Mod- #5: FS-iA6B - Failsafe Not Working When Using IBUS

Hands0n, 2016-01-30:
Hi, firstly I'd like to congratulate the developer(s) on this splendid enhancement to the FS-i6 Transmitter. Such a major improvement over stock.

The issue I want to raise is that, when using IBus to operate to Cleanflight/Baseflight the channel Failsafe does not function. In the event of a loss of signal (out of range or transmitter power failure) the channels remain at the setting they were at when the "failure" occurred.  

The effect is most easily viewed in the Cleanflight Configurator Receiver tab (see picture attached). 

This leaves a fairly high risk of fly away and total loss of the aircraft.

I have not tested the channel Failsafe with regular PWM use yet.  If you would like that done I would be happy to do so.

Can this be fixed?

[image]

Hands0n, 2016-01-30:
Attached another screenshot with sticks, switches, knobs set in random positions, remaining so after TX switched off.

[image]

Hands0n, 2016-02-06:
I want to cancel this Issue.  It turns out that I was making a rather fundamental error in setting up the Channel (3) FailSafe.  I have now been able to reliably demonstrate this working across i-Bus.  There is no issue.

closed 2016-02-06 by Hands0n

goldjacket, 2016-06-24:
I have the same issue.Except mine did fly away. How do you setup this channel 3 failsafe??? Im on a naze32 Cleanflight.
THanks

leonard4, 2016-08-08:
Also interested in how you set that up, seeing a similar issue with Failsafe set to OFF on all channels on my FS-i6, and Failsafe set to drop in my FC.

sebastianha, 2016-08-09:
I set one channel to rise it up to 100% on failsafe at the RC and configure the failsafe mode in cleanflight on this channel. This works very well for me.

goldjacket, 2016-08-09:
this method works well only if your receiver still has power or operational.  not sure why fc doesn't go into failsafe on a loss of ibus communication. that I think is what needs to be implemented. it's been a while but what happens is the ibus fc input thinks it's still at last known position and doesn't go to the internal fc failsafe.

fape, 2016-08-10:
After I set ch3 failsafe to -100% (others still OFF), cleanflight can go into failsafe mod when I power off my FS-I6.

mstanbis, 2018-12-07:
Been looking into the same issue myself and thought I'd post...even though it's been a while since the last one ;) 
From what I can figure out, IBUS or SBUS send a continual signal to the flight controller regardless of whether the TX is talking to the RX or not. Hence why a 'signal loss failsafe' condition doesn't occur in cleanflight/betaflight/iNAV. There's a few other ways of getting it done though. Here's a really good guide that talks further down the page about setting the TX/RX to send an "invalid" throttle setting through TX failsafe setup [https://www.propwashed.com/ibus-betaflight-guide/ ](url)
I came to a similar end stuffing about with the TX and set a spare channel to send a +100% value on failsafe of TX/RX then set iNAV manual failsafe flight mode to that value and channel. Here's some other info on YouTube from Paweł Spychalski https://youtu.be/Gh6i1kyxY6s .