benb0jangles/FlySky-i6-Mod- #6: Can't customize ch7 to ch10 in PPM

damdam2, 2016-02-04:
Hello. Congrats for all, I made the update very easily.
I bound the i6 moded with the i10 rx

I have a Naza FC and in PPM,  the flight mode channel is in ch7.
Channels 7, 8 , 9 and 10 can't be changed (reverse, endpoints, subtrim) so i can't customize them.

may the 4 last channels be customizable ? thank you in advance

leofaoro, 2016-02-27:
I have the same issue. I can program Ch 5 and 6, but not 7-10. Installed version from Jan 16/ 2016

srul, 2016-03-14:
Have the same issue...

cosy64, 2016-03-25:
have the same. and no channel 7-10 in the mixer

jarno83, 2016-04-22:
With ia6b i got PPM 6ch and ibus 10ch.

srul, 2016-04-22:
Can you please explain how you used the i bus 
What kind of fc are you using?

jarno83, 2016-04-22:
I'm using flip32(similar to naze32). Using latest cleanflight firmware. Follow sbus manuals but make sure you choose ibus from cleanflight configurator. And ia6b connect connect to top pins (ibus servo).

srul, 2016-04-23:
Thanks man
I don't understand spanish. But i got the idea 
I will try it today
Thanks agaim
Srul

jarno83, 2016-06-16:
Ok like I undersand then now we can have 8ch PPM with ia6b RX - http://www.rcgroups.com/forums/showthread.php?t=2677694

knt102000, 2016-08-05:
Any progress on this issue? Using NAZA with PPM for 8 channels, but because NAZA only sets flight mode on Ch 7, I can't set flight modes correctly unless adjusting endpoints is possible. 

I wouldn't mind taking a look at the code and messing with it if open sourced.

welshtom, 2016-09-19:
Is this possible at all? Would love to know

utsxumiao, 2016-11-14:
I am having exactly same issue on NAZA FC. Any chance get channel 7-10 endpoints setting? Anywhere the code is so we can mess up with?

kristoferahlqvist, 2017-06-07:
Same issue here. Would be great if I could use a custom mixer on channel 7

Kestov, 2017-06-17:
Same issue. Really sad to see it's a genuine problem, and not just something I overlooked.

givmen, 2020-01-31:
I have same issue. on naza v2. I am using tradition connection between receiver and fc. all 6 channels are busy. 5th on flight mode 6th on tarot gimbal. 7th in IOC but reverse is required on switch. I solved it to assign 7th chanel on variable A