benb0jangles/FlySky-i6-Mod- #10: End Points configuration somewhat broken on 10ch V1.5

danieltwagner, 2016-02-23:
I just set up the 10ch Firmware V1.5 and found that end point configuration doesn't quite work as it used to.
On the first attempt the cursor would jump down the left column, then start on the right column for ch1+2 and then move back to the left column for ch3+4 before returning to the right column for ch5+6. This would just repeat forever. I then set my ch5 right column to 60%, saved, and now my cursor ONLY goes down the left column, but at least does so in an orderly fashion. However, it never jumps to the right column.

Another user reported this for a (related?) controller a couple of years ago here: http://www.rcgroups.com/forums/showthread.php?t=1908088

Reproduces very reliably for me, please let me know if there's additional info I can provide.

ratozumbi, 2016-08-15:
It's not a bug.
It's the way that this menu work.
Left is minimum end point, right is maximum. You go from one column to another moving the stick of that channel from minimum to maximum.
Example: If you change from channel 2 to 3 and 3 is throttle at maximum (or any switch on), you will go from left to right.

**I also find that this calibration is really important to be done after flashing this firmware, because 100% of the end points doesn't correspond to 100% of the stick movement.**

danieltwagner, 2016-08-15:
Hum, maybe my description wasn't clear? I found that after the firmware update I can no longer reach all settings. Before I could cycle through all minimum and all maximum endpoints, but after the update it would cycle through a smaller subset. I would not able to e.g. set ch3 or 4 maximum endpoints, ever. I would be able to set: ch1,2,5,6 minimums and ch3,4 maximums. After editing ch5 max the cursor stopped jumping about and only cycled among the min columns of ch1-6, but that meant I can now no longer edit the max endpoints.

ratozumbi, 2016-08-15:
The menu you are talking about is functions > end points, right?
I have another controller here with Eachine stock firmware and it behaves
the same as the v1.5 of this project.

Try this: When you get to e.g channel 3 or 4 push the corresponding stick
to the maximum and see if it goes to the right column.
Also, don't use any flight controller to make this calibration. Use in
controller menu funtions > display
If that minimum or maximum is past some value, the end points menu might
get some wrong readings whetter you are pushing to maximum or minimum

danieltwagner, 2016-08-15:
Oh wow, you're right. The behaviour that you describe is indeed what's happening here.
Whatever firmware I had before (FlySky factory) would just cycle through all of the minimums first and then proceed to go to all of the maximums, which was much more discoverable... Anyhow, thanks for explaining this. I can set all my endpoints again! :)

closed 2016-08-15 by danieltwagner