benb0jangles/FlySky-i6-Mod- #13: I dont become view full file on my screen!

Holger71, 2016-03-22:
I want download the /10ch_MOD_i6_Programmer_V1_5.exe. But i hav no "view full file on my screen.
There is only "view Raw"! This opend a Window and i can start the Download.
I dont know if the Prgrammer is ok by getting on thes way.

Sorry for bad english!

benb0jangles, 2016-03-28:
It is just the way Github is. You have to learn.

Holger71, 2018-03-21:
Hello benb0jangles, the SW works well.
now i have one question.
is it posible to switch of the Protection for Modellswitch and for the Switchs and Sticks when turnon the Transmitter.
I dont need thes Protections because i drive Trucks, Weelloader and Traktors. 
I think there are many People with Boats, Trucks and so who need a Software without these Protections.
iI hope you can make a Software without these Protections.
i have great respect for your work!
Yours sincerely, Holger