benb0jangles/FlySky-i6-Mod- #14: Cvt01 voltage alam

zbrig, 2016-03-25:


zbrig, 2016-03-25:
One more thing... Big thank you for your work. ;)

RobbieHood, 2016-06-25:
I have this working on my transmitter with the voltage sensor. You simply make the battery a 9 cell ext lipo for a 3s. Adjust voltage levels to match 3s et voila.

RobbieHood, 2016-06-25:
So in the RX battery section, I choose 9 cells ext lipo and set the voltage to high: 12.2 all: 11.1 low: 10.8

ThaRaat, 2016-07-26:
Hi, thanks for the grate Firmware... But i get it the alarm not working for an external sensor. I use the avt-01 and the Voltage and the display shows the correct Voltage, but the alram doesnt work. Only for the internal sensor. :-(

RobbieHood, 2016-08-16:
Lol I feel so dumb. Only just found out last week that in the rx battery you need to set high voltage first then medium finishing with low. This way you can just set any cells you want and the correct voltages. All working perfect here now.