benb0jangles/FlySky-i6-Mod- #16: RE: Bluetooth link

inkyhacker, 2016-04-03:
Just found this it may be of help,
 http://www.rclogger.com/RC-EYE/Spare-Parts-Upgrades/Spare-Parts-NovaX/RC-Logger-Datalink.html
I have updated my radio and i must say, IT'S GREAT!!!

Thankyou! If you ever need help just email me ("Inkychips@gmail.com")

benb0jangles, 2016-04-04:
I saw that BT, you can make your own too :)
Does it work with APM telemetry? What receiver are you using?

inkyhacker, 2016-04-05:
It's on order, Ill update when i get more, Just had an idea though.

Could you implement multiple receiver support?? so when you switch models the receiver bind code also changes?? instead of only being able to have 1 receiver binded for all your models?

benb0jangles, 2016-04-06:
I hear that for multiple receivers, you have to copy the model to another free model location, and bind receiver #2. each model location only supports one receiver. I *think* from memory the i6 is this way,