benb0jangles/FlySky-i6-Mod- #17: RSSI signal from tx to rx possible?

Flo100, 2016-04-12:
Thank you for this reat mod!
I really like the 10 channels and got them working with i-bus and betaflight.
Is it possible to put the RSSI signal back into an aux channel?
I can assign all switches to the aux channels, but I would like to get the RSSI signal back to the receiver via an aux channel.
This way I can setup betaflight to show the RSSI on my osd.

marki555, 2016-04-24:
RSSI is the indication of how the rx sees the signal from tx. It then transmits it via telemetry back to the tx so you can see it on its LCD display. But the value originates on the rx, so there is no reason to re-transmit it back to the rx again :).
With little soldering you can get the RSSI value to OSD, for example like here: https://www.insidegadgets.com/2015/11/25/extracting-ia6-receiver-rssi-and-outputting-to-osd-for-fpv-display/

rootik, 2016-07-20:
I second that request. So no soldering needed at all on ia6b rx.

danarrib, 2017-01-02:
I'm endorsing this request. I have no soldering skills, so it'll be great if the transmitter sends back the RSSI to the FC using an AUX channel (Channel 10 will be fine).

D17E, 2017-02-09:
so....have anyone got solution for this Tx...?

alvaroloes, 2017-02-09:
I would love to see this coming true

rootik, 2017-02-09:
Take a look https://www.rcgroups.com/forums/showpost.php?p=36175778&postcount=1134
This modified firmware can transmit RSSI to RX as spare channel.
On GitHub: https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Timer%20Mod%20i6%20Updater/10ch_Timer_MOD_i6_Programmer_V1_4

MylesShannon, 2017-02-10:
Yep, it works for me. Set one of your channels under "aux channels" to "error" on your flysky. Then set the RSSI channel accordingly on your flight controller.

alvaroloes, 2017-02-13:
Hey! I didn't see that update. I'll flash my flysky with it and take a look. Thank you!

zelwake, 2017-02-19:
so i just need to set some channel to error and the same one on fc to rssi and it will work? does it nees some special wiring or ibus servo will handle this?

sebastianha, 2017-02-19:
Yes, just set one channel to error and set the channel in the receiver tab in cleanflight/betaflight accordingly. That's all. Note that the channels start from 1 and not from AUX1, so if you set first AUX channel you have to select channel 5 for RSSI when I remember correctly.

zelwake, 2017-02-19:
its really working wow, thanks, didnt think it would be so "easy" (since someone had to find how to mod it)

daxscooter, 2017-02-20:
Brilliant!!!!! Works a treat!!! Got rssi with no soldering and loads of extra functionality from my ia6b....... sbus options rather than ibus and a timer!!! Though how do you stop the timer other than turning off the tx?

Sbus pwm options are awesome! Allows for my camera servo (for different modes, I.e. Angle for take off and landing and then 45-60 deg inclinations for 2 rate modes) awesome!!!!

Thank you guys for this, have been looking for ages for an rssi solution and this rx seems to be the BOMB!!!!

I have de-pinned my ia6b, added a 5 plug connection and use ibus (now sbus with my NEW betaflight f3 FC) 5v, ground, signal, pwm for camera servo (Chanel 5). Wicked little set up and inexpensive! So much functionality!!!

Cheers for the work guys!!

Cheers, chris.

daxscooter, 2017-02-20:
Oh I forgot to mention the voltage telemetry mod which is the last of the five cables from my ia6b to battery +Ve for tx voltage reading!!!

All round great set up!

daxscooter, 2017-02-20:
For the timer, would be nice if it start / stopped when you did-arm / dis-arm. I have mine set to start when arm-on!! Any ideas?

rootik, 2017-02-20:
Guys, great to hear this RSSI mod success. Can you tell me, is it posdible to set error channel above 10? Because ia6b rx sending 14 ch PPM on i-bus.

daxscooter, 2017-02-21:
I see the 14 channels on betaflight, but at the moment tx is only putting out 10 channels on 'aux channels' screen. Anyone fancy updating the firmware to 14 channels from the rx? Would free up the 10th usable channel from the fs-i6!?!

daxscooter, 2017-02-21:
I mean updating the No of channels from the tx! I.e from 10 to 14?

lovejoy777, 2017-02-23:
working perfect thank you

alvaroloes, 2017-03-21:
I finally got the time to flash my FlySky and it is working like a charm. Thank you very much.

MathijsG, 2017-05-03:
Can people elaborate on how to set this up in transmitter? Where can I map error to a channel? Can't find it.

zelwake, 2017-05-03:
@MathijsG Functions ->Aux. channels -> pick one (usually channel10) and set Error

MathijsG, 2017-05-03:
Don't have the option here to map error to it. I'm using this firmware: https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Mod%20i6%20Updater/10ch_MOD_i6_Programmer_V1_5 it should be the latest. What am I doing wrong?

zelwake, 2017-05-03:
yeah, that doesnt have it, try this one: https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Timer%20Mod%20i6%20Updater/10ch_Timer_MOD_i6_Programmer_V1_4 since v1_2 it has error (rssi) as aditional function but this one fixed timer offset or something that was wrong

MathijsG, 2017-05-03:
So the older one, 1.4. is the better one? Does it still have all the features this original custom firmware has?

zelwake, 2017-05-03:
actually, this 10channel_Timer firmware is newer than just the 10channel firmware. 
and tbf I never updated firmware on any of my receivers and it works just normal

alvaroloes, 2017-05-04:
@MathijsG In case it helps, this is all I did: 

1. I updated the transmitter firmware using "10ch_Timer_MOD_i6_Programmer_V1_4" following the instructions in the Readme. Everything worked flawlessly: I had 10 channels available and I could set channel 10 to "error".
1. Then, in Betaflight, I went to the transmitter tab and set the RSSI channel to 10. Take a look at the screenshot:

[image]
 
After these steps, I could see the RSSI in my OSD 😃

MathijsG, 2017-05-04:
Thanks mate for your explanation :-). Will try further tonight to update the firmware. I still don't understand why this timer firmware is a different firmware than the regular. Why did they split them anyways? I think I can't backup my TX settings, so I will write everything down and flash the transmitter.

I still can use IBUS/SBUS as communication type in betaflight? Now using ibus/sbus, not ppm.

alvaroloes, 2017-05-04:
You're welcome. Yeah, I don't understand the split either...
Regarding your other questions: Although it is a good practice to backup (somehow) your TX settings before flashing a firmware, all the settings remained after the upgrade (at least in my case). And yes! I'm using IBUS without problems.

zelwake, 2017-05-04:
Because it was a new project that still has some bugs (on mine I cant get to secret menu, not like I ever need to) while the original 10channel firmware is finished and clean.

MathijsG, 2017-05-04:
I have it working now, technically. But I see it as 0%. Instead of 100%, or is it still the error rate and not the actual RSSI? So 0 is good? Does anyone know the settings for minimosd for it? Do I need PWM enabled RSSI or no PWM?

I still can get in factory menu with this firmware. Although I can't select other sensors to be on the main screen. On the other firmware you can see the error rate and the RX voltage (although it's always 5 volts)

zelwake, 2017-05-04:
Yeah, lower is better so just error rate. For OSD I believe you have to use FC with OSD built in (betaflight, omnibus).  

I can, just have quad powered and it allow me to set any sensor I have set on it (battery voltage, error). As for the RX voltage you see, thats power that your receiver is getting from fc which is always regulated to 5V, you need to use vbat on x6b or either modding ia6b or using voltage meter that is sold separately.

MathijsG, 2017-05-04:
Yeah I mean, I use MinimOSD chip with MWOSD in it. To display RSSI I can setup some parameters:

RSSI
Display RSSI - select this if you wish to display RSSI on the OSD.
If neither FC or PWM are selected, the OSD will use ANALOGUE RSSI from a RX directly connected to the OSD
Use FC - select this if you wish to to use the RSSI value from the Flight Controller instead of the OSD. The FC must configured to support this
Use PWM - select this if you wish to use the PWM RSSI value from a RX connected directly to the OSD
Use PWM - (with RCRSSI configured in config.h) select this if you want to use a RC channel 0-7 for RSSI
RSSI alarm - set this to the value at which you require the RSSI value to start to flash
SET (Min) - click this to calibrate RSSI minimum value with TX turned off
SET (Max) - click this to calibrate RSSI maximum value with TX turned on
!! NOTE !! See guide for how to calibrate
Source: https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/User_Guide.md

Im playing with the settings. But I'm supposed to see 0% as a value? Because RSSI is normally inverted 100% is 100% quality connection and 0% means no connection.

zelwake, 2017-05-04:
dunno, i had only osd on pdb and cam and none of them can do rssi, cant help you here

MathijsG, 2017-05-04:
Where do you use your RSSI then?

The channel which I map to 'Error' in the transmitter always get's a value of 50% in Betaflight.

I will make pictures of where I'm stuck tomorrow.

MathijsG, 2017-05-14:
This is the problem:

I mapped the error to the 5th channel:
[image]
[image]

This is the default value of the ERROR: 
[image] 
[image]
(with 0% error rate)

Here I can only change the outer 100% value, not the first 100% at channel 5:
[image]

Do I need this enabled?
[image]

These settings I can alter in my OSD:
[image]

So the questions are:

What is the range for the ia6b in error terms? By default it's about 2000 for 0% error rate.

dml51, 2017-05-19:
Sounds like a great mod, does the dev have any plans on making an i6x version? Of course that's what i own, so i can help if needed :)

qba667, 2017-05-29:
@MathijsG 
1.100% error the value is send as 1000, 0% error value is send as 2000
2,3 It is pure digital value - I was using error channel in cleanflight. 
With latest FW error is also send as channel 11.

@dml51 - I don't have i6X so i would say no... but if FlySky will play nice we can get a way to completely open FW.

MathijsG, 2017-05-29:
So the value of error is lower than zero error. I will try to figure out if my OSD can handle that, because often it's the other way around.

qba667, 2017-05-29:
@MathijsG Error value send as channel data is reflecting connection quality let say similar to RSSI.
It is calculated with formula:
1000 + ((100-error)*10)
So relations are following:

100 = 1000
90 = 1100
80 = 1200
...
20 = 1800
10 = 1900
0 = 2000

MathijsG, 2017-05-29:
I will puzzle with this soon. Hope I can somehow communicate this scale and setting to betaflight so that it will understand the RSSI value.

MylesShannon, 2017-05-29:
@qba667 
Link us to the latest firmware? I'm not seeing an update that puts RSSI on channel 11 with FlySky-i6-Mod. I'd sure like to use VrB on channel 10 again.

MathijsG, 2017-05-29:
Maybe this firmware? https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20qba667_hali9%20Updater/10ch_qba667_hali9_i6_Programmer_V1

I don't understand why there are all this parallel firmwares for the exact same device. It's quite hard to understand which version to use.

qba667, 2017-05-29:
@MylesShannon take always latest from https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch!/page129
But look at comments yesterday I have attached some old firmware for ALT sensor test.
I am not an admin of this repository, so I have no impact on what is published. 
@MathijsG have you tried this method:
http://djwoodz.com/2016/04/14/Betaflight-RSSI-on-MinimOSD-over-SBUS-X4R-SB/
There is no need to scale the value in TX or Betaflight it can be done in MWOSD by setting 
SET RSSI MAX: xxx
SET RSSI MIN: yyy

MathijsG, 2017-05-29:
I finally got it working earlier today. I think I also mapped the channel wrong. I was looking at the AUX number as the channel number, but it's the raw channel number where the movement Axes also count in. 

I will add some clear documentation for people, because you can easily do something wrong. Now I've got RSSI in both my goggles as well as in my Flight Controller :-).

qba667, 2017-05-30:
@MathijsG great - unfortunately I have no time to document all what have been made. Can you test for me what is lowest possible error rate what we can use to fly?

MathijsG, 2017-05-30:
I have set the RSSI_SCALE to 2, but I will play with it more later. The main confusion was I think that Betaflight has named the channels AUX 1 - 12 and excluding the control channels in this range. So I thought when I had RSSI on AUX channel 5, I needed to pick Channel 5 from the RSSI channel list. However this list DOES add the control channels, so AUX channel 5 would probably be channel 9 (because the first 4 control channels are added)

About your question, I can try a little, but I don't want to crash with my quad :-P. And the decrease in range isn't lineair, but sometimes randomly fluctuating.

alias1atl, 2017-08-30:
PLEASE HELP! I have the "TGY-i6S" & would like to add the error AUX channel. so I can add RSSI to my OSD. I already have 10 channels & the "10ch_Timer_MOD_i6_Programmer_V1_4.exe" doesn't load because its for the i6 not the i6s.  Also I do have voltage & signal (RSSI) on my TX screen without any mods but I want them in my OSD.  Dont want to take goggles off to see if I have a wake RSSI signal

MathijsG, 2017-08-31:
If you can't flash the firmware you can't have the RSSI in goggles. Simple as that. With the stock firmware you can't map the 'Error' read outs to a certain channel. With this custom firmware you cán.

BASSBEAT4LIFE, 2017-08-31:
First THANKS to the creator of this mod it is great BUUUUUUTTTT i cannt bind my quad with it -.- so all the good stuff is sensless for me -.- 

I switched in to AFHDS prot cause this is what my flight controler use.... I did the binding like i always do... set the radio on bind, set the drone on bind, switch of the radio, than the drone and after restarting the controller its bind... but after this mod nothing happens.... i flashed back to a older 10ch mod (but less funktions....) and there i get the bind really quickly.... i also flashed back to stock and tried there an also with that software the binding wos easy but i'm not able to bind with this mod -.- I use a Furibee F3 32bit board

Please please help me i whant to use THIS mod !!!!

MathijsG, 2017-08-31:
@BASSBEAT4LIFE  Please file a separate issue for problems not related to this thread.

nowayhaze, 2017-12-22:
@alias1atl I'm in the same situation but with an FS-i6S. Did you ever figure it out?

I saw that there might be a way to solder a wire onto the receiver to get the RSSI signal out, but I don't trust my soldering skills on such delicate and small parts in my FS-iA6B.

If only the custom i6 firmware could be ported to i6S. Does anyone have any info on this?

syokonline, 2017-12-28:
Hi im already use error value to my minimOSD. That is actual value i can compare between flyskyi6 and at on screen display but need to adjust untill match.. here the video https://www.youtube.com/watch?v=b6Xz68SzQYA