benb0jangles/FlySky-i6-Mod- #18: Throttle warning at boot.

jarno83, 2016-04-16:
Hi, the 10ch ver1.5 firmware flashed fine but at restart I get Warning that Throttle needs to be at min and switches up, but what ever I try I can't get past that Warning. Any thoughs? Thanks

penner180, 2016-04-16:
Try starting in hidden star menu put both sticks bottem left. Then turn
controller on... You should be able to relearn movement on the controles

jarno83, 2016-04-17:
If I flash orig. firmware all works and works with Nova firmware also but endpoints and center points are way off. Can't go to secret menu also :(

jarno83, 2016-04-17:
Ok, did some more digging and my TX only works with Original FS-i6-Update-20151015 updater and RCLogger Nova X350 firmware. Any other Uploader and firmware gives Throttle warning and no secret menu also.

jarno83, 2016-04-20:
No one has same problem?

jarno83, 2016-04-22:
What if I unplug the gimbals and switches from mainboard... can I move past the throttle warning then? Maybe can somehow to enter secret menu and calibrate sticks. I see that with novax350 firmware they are off from normal like throttle min is 1150 or so. After calibrating it's ok. What do you think? Thanks.

jarno83, 2016-04-22:
Ok, got it working. The problem was with throttle pot. The value didn't go low enough, when turning it by hand all worked. The gimbals are cheap so there ain't any adjustment screws and I had to cut a 1-2mm off the plastic stoppers and Voila :)

closed 2016-04-22 by jarno83

seregio12, 2016-04-27:
i have the same problem, i flashed the 10ch ver1.5 firmware and i have swtiches and throttle warning, what should do i ?
thanks.

jarno83, 2016-04-27:
Hi, check mu last post, there I explaned the problem I had and a quick fix for it. Good Luck

seregio12, 2016-04-27:
Yes, i saw but i do not understand how you cut the stoppers. where are them ?

seregio12, 2016-04-27:
I got it. i opened the tx and i cut plastic stoppers and it works. 
i do not understand that before it worked and after update firmware it failed.
thanks so much.

jarno83, 2016-04-27:
Seams like the official latest 1.1 acts a bit different with pot valus as modded firmware. Glad to help😀

openscript, 2016-06-24:
Does anyone has a copy of the FS-i6-Update-20151015? It would be great if the one could upload it somewhere, due their servers are down and I couldn't find it somewhere else on the net. Thank you in advance!

sampao25, 2016-09-01:
Hello I'm sorry as don't quite understand where the plastic stoppers are located inside the TX.  Would someone be able to upload a picture?

Thanks in advance!

Appolloscott, 2016-09-02:
Hello, I am trying to set up my 10ch. Mod i6 to  work on a Flybarless helicopter. The Flybarless controller has CH mixing already built in; so I can't use the radio's swash mixes , because they will interfere with the FBL unit. I can however  use the presets for a "variable pitch helicopter " because it has no mixing. Unfortunately  the collective CH #6 does not work with the throttle CH # 3 like the setup for the other helicopters in the radio. I have found a fix for this by slaving CH 6 to CH 3 so that whenever I move the throttle the collective will also move.  My problem comes when I hit the throttle hold switch. The radio ignores CH 3, but because CH 6 is slave to it the collective is also ignored. So instead of throttle hold just stopping the motor I also have no collective. Does anyone know of a way to tie the throttle and collective so that one will work while the other is on hold. Tried every setting I could find with no luck.

fape, 2017-02-25:
I got the same. It works with offical firmware, but I got "swith and throttle warning" on boot.
Is it not possibly to fix with out disassably the transmitter and modify the hardware? Why the hardware modification is necessary?

mattyfreebles, 2018-03-08:
how can you revert to original firmware

fape, 2018-03-09:
I [downloaded](http://www.flysky-cn.com/downloadsfront.do?method=picker&flag=all&id=4589c98c-f03b-4f51-a319-61390264de0c&fileId=24&memstate=&v=2.zip) from official site

Boberling, 2018-06-05:
Well anyone know what is wrong if cutting stoppers and using whether FS-i6-Update-20151015 or novax350 would not help?
Because it would not for me with any firmware - only warning and no Factory meny(

dinho-limno, 2018-08-12:
Yes! wire jumped the swith B and iw works!

kevinmcguigan, 2019-03-12:
How do i get my fs t6 controoler past the warning screen

kevinmcguigan, 2019-03-12:
Anyone out there that can help a guy out?

fape, 2019-04-20:
I got a special version of the firmware where warning screens were removed.

siawzen, 2019-04-26:
anyone have ORIGINAL FW FS-i6-Update-20151015.exe or higher please share it, because official website is not found, Thanks

fape, 2019-04-26:
Here you are: http://bit.ly/2Le5XKq

sergeolkhovik, 2020-05-11:
I was able to handle that without any cut of plastic or so.
Simply launch factory menu (I was able to do that with both sticks in top-right corner, you may check proper position in display menu) and use 'sticks adjust' command, like here: 
https://www.youtube.com/watch?v=FkJRcjszEd8 

Strange, I thought initially that this warning was related to this firmware, but after I reverted back to original 2.0 I still had the same warning. So only that adjustment worked me.