benb0jangles/FlySky-i6-Mod- #23: Moded FlySky-i6 = 10 PPM Channels on the FS-iA10B?

tshaw1, 2016-05-22:
A simple question. I love the idea of the 10 Channel Mod that you have created, I am running that modification on my FlySky i6 Transmitter! And I understand that there may be hardware restrictions when trying to receive 10 PPM Channels via the FS-iA6B Receiver... However can anyone tell me that if I purchase the FS-iA10B Receiver, could I then utilize ALL 10 PPM Channels? I attempted to use Channel 7 (B Switch) with my Jiyi P2 FC to adjust flight Modes using the FC-iA6B Receiver and one cable running from Channel 1 to the RC port on my FC. It did not Recognize Channel 7.

arun2118, 2016-05-23:
Yes, your able able to use all ten pwm out channels on ia10 receiver.

tshaw1, 2016-05-23:
I'm interested and need to utilize the PPM Protocol. Ironically, I've found since asking the question earlier today that even the rather new i10 Transmitter will only Transmit 8 PPM channels.

piodabro, 2016-10-26:
All FS-iAxx can provide only 8 PPM channels. Better solution is to use iBus where 10 channels are present.

dimak771, 2018-04-02:
Hello! Can I use 10 ch with IBUS + FS-iA6B Receiver? Not PPM.