benb0jangles/FlySky-i6-Mod- #25: Add audible warning when error rate is to high

sebastianha, 2016-06-02:
Would it be possible to add an audible warning, like the low battery beep tone, when the transmission error rate is too high?

chriscamacho, 2016-06-02:
just a thought actually seeing the error rate (if that's even possible) could be used as a crude model finder, would be worth a try with a foil "wave guide" it limit its field of view....

sebastianha, 2016-06-02:
By default you see the error rate on the main screen at the bottom. But when flying with goggles on an audible alert would be nice.

chriscamacho, 2016-06-02:
.... definitely worth an experiment!

Bahecuk, 2016-06-03:
Definetely worth a shot - just today I crashed twice due to weak RC signal, which I obviously cukdn't notice wearing FPV mask

cmtemantovani, 2016-06-03:
this guy put a LED to warn the signal
if you change, its possible to put a buzzer

https://www.insidegadgets.com/2015/09/03/adding-signal-strength-rssi-led-to-the-turnigy-tgy-i6-transmitter/

To put rssi in osd:

https://www.insidegadgets.com/2015/11/25/extracting-ia6-receiver-rssi-and-outputting-to-osd-for-fpv-display/

sebastianha, 2016-06-03:
I also saw that, but this requires hardware modification.
A buzzer (for battery warning) is already built-in and the signal is also there, so it should theoretically be possible to add this feature with a firmware modification only.

crozone, 2016-08-15:
There's already a low battery warning built into the firmware - we could reuse this for the RSSI warning? Anyone know how the state is managed for the alarm? I'm assuming there's a simple check on the battery levels every now and then which kicks off an alarm state machine.

johnryanlee, 2016-10-10:
I'd like the tx error audible warning as well. Has anyone managed to get the code for this yet?

batata004, 2017-11-13:
Hey, any update into this? Is audible rssi in the buzzer of the radio already available in the new firmware?

qba667, 2017-11-13:
@batata004 sure it is. 
Just configure RSSI/Error alarm using firmware https://github.com/qba667/FlySkyI6

batata004, 2017-11-13:
@qba667 thansk man! Just one question:

I see there is this firmware -> https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Timer%20Mod%20i6%20Updater/10ch_Timer_MOD_i6_Programmer_V1_4

Many people use that firmware which looks the latest.

So what is the difference of it to the https://github.com/qba667/FlySkyI6 ? Does https://github.com/qba667/FlySkyI6 is more updated (it looks cause it was updated at July).

Which one should I use to keep my radio as updated as possible?

qba667, 2017-11-13:
@batata004 I have no rights to this repository - firmware on this repo is my older version - with few bugs and gui updater. My repository contains more recent version and console updater.

batata004, 2017-11-13:
@qba667  great my friend! My TGY-I6 radio is with the original firmware of the factory. So I checked your repository https://github.com/qba667/FlySkyI6 and it is much more updated BUT I cant find the .exe file so I can run it on my windows while having my radio connected to usb.

Would you mind telling me how to flash you firmware in my radio? Cause the other repository (which looks the "oficial") provides EXE file to flash. Thanks a lot.

batata004, 2017-11-13:
@qba667 I will install this firmware of yours -> https://github.com/qba667/FlySkyI6/releases/tag/1.6.3

This looks the most recent, right? I will use arduino to flash it but I cant find the exe file, could you please point me in the right direction?

batata004, 2017-11-13:
I must be very stupid, inside the zip file there is a EXE file. But I cant find instructions how to do that.

qba667, 2017-11-13:
@batata004 just wire unit correctly, enter FW update mode and start the exe. 
Fw update mode trigger is explained on this repository.
You have to use NO_SWE version if you have not modded the hardware for additional switch.

batata004, 2017-11-13:
@qba667 right. I never moded my radio, never installed a firmware on it. So I need to use the file "NO_SWE_14_channels_flysky-updater_x64.zip" right?

I will enter in FW update mode and execute the file! Just one final question: if I ever want to come back to the current firmware that I already have in my radio, how should I do that?

qba667, 2017-11-13:
@batata004 Then use official updater from FlySky:
http://www.flysky-cn.com/download.html

batata004, 2017-11-13:
@qba667 I just installed your most recent firmware in my TGY-I6 and it worked but somethings dont look right.

1) the voltage RX alarm is with some problem, I set 3 cells and set the high to 12.6, alarm to 11 and low to 10.5 BUT even when 11v is achieved the alarm beeps for a while and after that stops even with my battery decreasing it's voltage. And even when it reaches 10.5 the alarm does not beep;

2) using the ia6 receiver the radio DOES NOT show the DB (RSSI) value. With the stock firmware it would always show the RSSI value. Why this firmware does not do that?

3) when I enter in some menus EXTRA -> Timer (for example) all the fields are filled with random very high numbers (not zero). Ok I can lower than but that's really odd;

4) every 1 or 2 minutes my radio beeps a LOW - HIGH tone. I cant understand why it's doing that (I am not leaving the radio stopped, even using it this beep keeps happening and I think this beep is cuting the battery alarm beep and it wont stop anymore).
 5) when setting some values (using UP/DOWN) in RX -> RX BATTERY -> there are 3 rows (LO, AL, HI). For some reason the value in LO always finished in 8, the value in AL always finishes in 3 and HI finishes in 6. Even moving the UP/DOWN it will always keep these values at the end.

For example: 

LO 10.58
AL  11.03
HI 12.67

I think this firmware you provided is mostly to be used in the FS-I6 radio. I know TGY-I6 is very similar but the odd things above looks weird.

I am doing something wrong my friend? If you need a video I can post!

batata004, 2017-11-13:
Also, I couldnt find the RSSI alarm which is the reason why I decided to install a custom firmware :(

batata004, 2017-11-13:
One more thing: suppose I configured my lipo alarm at:

LO 11
AL 11.1
HI 12.6

When the voltage reaches 11.1 the beep start playing. Ok, great. BUT when the voltage reaches 11 or below no beep is played anymore, nothing. In the STOCK firmware the beep would play at AL voltage AND AT LO VOLTAGE it would beep a much more persistent beep.

But in this firmware the behaviour is different, at LO voltage it stops completely beeping!

qba667, 2017-11-13:
@batata004 
1. I have to check this one but I never changed this part before - this was oryginal implementation by Dave - possibly is buggy - must be checked.
2. Check if it is on sensor list - nobody reported it before. Then configure it under ALARMS.
As far I know FlySky unit shows only Error rate RSSI was not parsed - what version of FW were loaded before?
3. Long press OK resets currently selected value - mod uses before configured  EEPROM area - there is no way to check if this is first run to reset values automatically.
4. Use Tx battery - the alarm voltage must be changed there
5. Try to reset values first. Then radio distinguishes long and short press.

batata004, 2017-11-13:
@qba667 thanks for your help! I rewrote my comment so it gets a little bit easier to understand ok?

I rewrote mainly the item 1), could you please simulate this bug in your radio? I usually set my battery alarm very close to he low level cause I would like a more intense beep at low but it's not happening.

About the sensor list I cant understand how it works... I made available the ERR% sensor now (I didnt now I should do that, thanks again!) and in the sensor list I cant find a way to remap things. Is there a manual saying somewhere how it works?

OBS: the long press of the OK button worked fine :)

batata004, 2017-11-13:
@qba667 I think I managed to understand sensor! But I think there is one problem: when I go to sensor there are 6 slots. If I set ERR to 4 or 5 or 6 it gets displayed in my screen BUT if in the 1, 2 or 3 slots no matter what I select it never appears in the screen. Why?

qba667, 2017-11-13:
@batata004 
I will try to reproduce the issue from point 1.
RSSI and SNR are available from FW 1.6 of IA6B - before only ERR(or) rate is provided. 
In Extra Alarms the alarm can be configured for telemetry value for example 
ERR > 50

There is more than 1 main screen - just press UP/DOWN when main screen is being displayed.

batata004, 2017-11-13:
@qba667 AMAZING!!! I can indeeed press UP/DOWN in the main screen!!! IT worked!! Now I got things to be displaid correctly!

I also set the alarm to ERR > 50! :)

Will be waiting you to take a look at point 1) I think this is a important thing.

One final question (I promise): the EXTRA menu is not saved to each profile I set in my radio, right? I mean, I use lots of different models profiles in my radio and I am realizing the EXTRA menu is global, it's not saved separetly with each model data, right?

qba667, 2017-11-13:
@batata004 for now it is global - but I have started work to make it per model.

batata004, 2017-11-13:
@qba667  thanks so much! That's a great firmware, good job man.

qba667, 2017-11-13:
@batata00 They are 2 different types of beeps - alarms are triggered correctly.

batata004, 2017-11-13:
My friend, I am facing some bugs and weird things with your firmware.

For example, now that I paired another receiver my radio does not display anymore the RX voltage. It was working fine before and now the IntV1 is not available anymore in the sensor lists. I tried changing model, factory reseting... nothing works.

Do you know the most stable version I could try? Maybe https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Mod%20i6%20Updater/10ch_MOD_i6_Programmer_V1_5 is the latest stable version?

qba667, 2017-11-13:
@batata004 This version has binding bug - but it has fewer changes. Please read rcgroups topic https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch%21/page176

batata004, 2017-11-13:
My friend, it looks like you firmware for some reason disabled telemetry of my receivers! It was working perfectly, I just added a new receiver and the telemetry stopped working in all my model's profiles. Any idea what might have happened? I just updated the firmware of my radio, I didnt touch the receiver's firmware. Any idea what might be happening?

qba667, 2017-11-13:
Please try different firmware - there is no control over telemetry stream - we are always parsing what we get from RX. You can try original version - just observe differences. If you will be able to find dependencies and write me how to reproduce it everything can be fixed.

batata004, 2017-11-13:
@qba667 first I want to apoligize if I am being disrespectful, you are very kind and you are helping a ton!

I am just sad because I really liked your firmware (I liked!) but I am having weird problems. For example, I tried now flashing the original firmware available at https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/Firmwares/Flysky%20i6%20Original%20Firmware but for some reason my radio cant still read telemetry anymore! :(

You know what is even stranger? When I bought my radio it came back with Turnigy firmware installed but the link above makes my radio display FLYSKy brand. Do you know where I can really find the original Turnigy firmware (not FlySky one) so I can try and check if telemtry is working?

batata004, 2017-11-13:
Please my friend watch this : https://photos.app.goo.gl/RlWNFsBffq0VrroZ2

qba667, 2017-11-13:
@batata004 Can you verify that connection is still established? On your video it look that there is no link at all.
Use exe file located here:
https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/Firmwares/Turnigy%20i6%20Original%20Firmware

qba667, 2017-11-13:
@batata004 are the RX units also TGY branded?

batata004, 2017-11-13:
@qba667 the file at https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/Firmwares/Turnigy%20i6%20Original%20Firmware also display FlySky logo/name on the display. 

My RX ia6b and ia6 are TGY branded. They were working just fine before updating the firmware.

I think I found a bug: when I added a third receiver and changed the receiver of my model #1 to the #2 the problems started happening. Have no idea why. So I flashed again your firmware and did a factory reset (in the SYSTEM menu) and after that everything worked perfectly. Now the telemetry is working fine. I have no idea what happened, but I think a factory reset should be mandatory when upgrading from another version of firmware! This should be stated at the readme file!