benb0jangles/FlySky-i6-Mod- #28: 10ch_MOD_i6_Programmer executable does not run

codeMonkeyDo, 2016-06-06:
I an running Windows 10 Pro, 64-bit. The 10ch_MOD_i6_Programmer executable does not run from any of the version directories. 

Attempting to run the executable causes a MSCOMM32.OSX file to be generated in the local directory (on disk size: 104KB).

Attempting to run in compatibility mode sees no success. 

Also, where is the source code for this project?

NicHyslop, 2016-06-07:
Try running the executable by right-clicking it and then selecting "Run as Administrator". That fixed the issue for me.

codeMonkeyDo, 2016-06-07:
That was the problem - thank you. 

I checked the readme and I couldn't see this mentioned. Why does it need to be run with Admin Privilages? 

It's other wise ran and installed with no problems. 

I did notice that the modified firmware is based of a 2014 build, my Tx had a more current 2015 version on it. But that's probably another issue. As it stands it seems to have worked! Thank you to the dev's!

codeMonkeyDo, 2016-06-07:
I had closed the issue but have now decided to re-open as I think the requirement to run as admin needs to be either addressed or documented. 

Maybe it is documented somewhere and I just missed it?

closed 2016-06-07 by codeMonkeyDo

NicHyslop, 2016-06-07:
I didn't see it documented anywhere; I discovered this while watching a YouTube video actually!