benb0jangles/FlySky-i6-Mod- #32: Error by Heli Mode variable Pitch - VRB status "used"

hagenp, 2016-07-14:
Hallo benb0jangles,

thank you for 1.5 Beta. It’s a really great work and helps a lot users for more modifications!

Unfortunately I’ve established an error, if used Heli mode with variable pitch by channel 6.  Normal in this mode should work channel 3 and 6 automatically together, but they are full disconnected. 

Therefore it has also no consequences if any values changes in the setting point pitch diagram. This function is urgent, when someone want to fly heli in normal and in 3D mode with variable pitch. 

Could you please check this error and correct betimes, 3 x please :-)?

Perhaps in this context it is also possible to decouple the dial VRB status "used" in “variable pitch mode”, thereby it is then possibility, to adjust via VRB the pitch with a mixer.

If you can this integrate in you new update, this would be realy cool.

Regards
P. Hagen

Appolloscott, 2016-09-04:
I had the same question

VAL72, 2016-09-30:
Hi,
please you did a great job, but we need help for 3D Helicopters. I checked it several times in the radio. Ch6 is directly connected to VRB. There is no asignment of Ch6 to the thottle curve and it's adjustment (or it is optional hidden somewhere). adjusting the trottle curve is influencing the collective pitch at ch3. What does the Pitch curve setup? So the thottle curve setup is at the moment the function of Pitch Curve, but it should be assigned to Thottle not to collective Pitch. My understanding for Thottle curve for 3D Heli would be: Left Stick Up/Dn at mode2 setup mixing via trottle curve via VRB level to Output Ch6. And also as implemented Left Stick Up/Dn at mode2 setup Collective Ch3 transfering via Pitch Curve (not thottle curve). So we may separate adjust The Heli Mode Normal and Idle Up as it is normal used in other Radios. This would be great for 3D Heli Motor. Isn't it?

Kind Regards
VAL

Appolloscott, 2016-10-07:
Same here thanks for all your work. Wanted so bad to use your update but could not. I fly heli and the CH 6 issue makes it impossible to operate a collective heli safely. Thanks again for your work and hope to see more updates.

qba667, 2017-02-26:
@Appolloscott @VAL72 please see 
https://github.com/benb0jangles/FlySky-i6-Mod-/issues/64