benb0jangles/FlySky-i6-Mod- #34: 10ch mod issue

afonsus1997, 2016-07-19:
Everything is working except when ch10 is on channels 1, 2, 3 and 4 doesent work. This can be checked on the display option.

lrusnac, 2016-07-27:
it works for me

ratozumbi, 2016-08-15:
it works for me

RobbieHood, 2016-08-16:
Same here. When switching switch d it gives throttle and such because the sport mode need to be assigned to a switch. These modes should have a "none" option to cancel it.

RobbieHood, 2016-08-16:
I use switch A as arming switch, switch B+D for a sic position flight mode switch and all default modes mapped to switch D so I dont accidentally switch it. Because sport mode is on the switch D it will give half throttle by itself and response in throttle just drops out.

Firmware needs to be modified like said above to have a "none" option available instead of the abbility to only assign switches to sport mode,throttle hold etc...

CFitzsimmons, 2019-09-12:
I also have the issue where SWD causes the throttle to move.  In the "function->switches assign" menu I have the setting for "Thro hold SWD" set to OFF.  However, it still affects the throttle.  To get around this for now I turn it to ON and in the "throttle hold" menu I set the value to 0%.  For now I just use this switch for my buzzer, but it can't be used for anything during active flight because of this issue.  Currently running "10 ch Timer Mod" version 1.4.