benb0jangles/FlySky-i6-Mod- #37: not opening

aryangangurde, 2016-08-18:
Please help, I have been at this for 5 hours straight and my head has no more hair to pull out... the file downloads but nothing happens when i click it

piodabro, 2016-08-18:
Open as administrator...

appletaartje, 2016-08-23:
Please help me, I can not seem to open the .exe files (e.g. FS-i6-Original-Firmware.exe) on my laptop. 
On Windows 7 64-bit I get an error that the version of this file is not compatible with the version of Windows I am running. Trying to run the file in compatibility mode does not help.. 

I also created a VM which runs on Windows XP mode (32-bit), but here I get another error which says "16 bit MS-DOS Subsystem" and "The NTVDM CPU has encountered an illegal instruction. 

Googling is nice, but give way to many options to solve it, and the most common solutions do not work (booting in safe mode or clean mode e.g.)

Please help me with this issue, because i would like to transfer my TGY-i6 to 10 channels.

Thanks so much!!

Erik.

piodabro, 2016-08-23:
FS-i6-Original-Firmware is not the 10-ch version...

Use this one: https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Mod%20i6%20Updater/10ch_MOD_i6_Programmer_V1_5

Remember to run .exe as administrator (Right click and choose Run as administrator)
For me on Win 10 64-bit after double click nothing happened. Running as administrator does the job :)

FS-i6-Original-Firmware is original firmware flasher (6-ch).

appletaartje, 2016-08-23:
Thank you Piodabro, of course I have already tried to run as Administrator, as you posted. But that did not help. 
I also know that the 10ch_MOD_i6 exe is the one to use, but none of the exe files seem to run.
The 10ch_MOD exes give the same error.

Also the errors that I get are not the same as yours.

Thanks.

piodabro, 2016-08-23:
Try to re-download exe's. Mine worked good in Win 10 64-bit. Your errors probably says that the file you try to execyte is corrupted.

appletaartje, 2016-08-23:
Redownloaded, same issue.... 
Antivirus says nothing, plus I already tried to execute without antivirus running.

So... im out of options. :(

appletaartje, 2016-08-28:
Any news on this topic??

Can I ask how large the files are that you guys have downloaded (name + size)?

appletaartje, 2016-08-28:
OH
MY 
GOD

Can it be so simple?? Why isn't there any manual on how to do this? People that don't know Github will probably be stuck for ages, because just clicking on the file (.exe) will only download the small version of the file (44KB) and not the "full" version (multiple MB's).

So, for me the simple solution is:

1. **Leftclick** on the "folder" in which you want to download a file (e.g. "10ch Mod i6 Updater").
2. **Leftclick** on the next folder (e.g. "10ch_MOD_i6_Programmer_V1_5")
3. **Leftclick** (not rightclick save as....!!) on the .exe file (e.g. "10ch_MOD_i6_Programmer_V1_5.exe")
4. A new page will be loaded and will show the real size of the file (e.g. 2.01 MB)
5. There is a **BUTTON** "_Download_". **Leftclick** on it......

Voila, file downloaded correctly and can be opened.
The other way around (the 44KB file) will not work, and gave me errors like: The version of this file is not compatible with the version of Windows you are running. And also like: NTVDM CPU Has Encountered an Illegal Instruction.

So this is what worked for me, nice to have found out myself, for nobody on this site could help.........

Cheers!

alan228, 2016-09-01:
Thank you so much for figuring this out "appletaartje" no one bothered to mention how github.com works but thank you I've been boggled over this for over a month and now it works perfectly

aryangangurde, 2016-09-01:
Have'nt confirmed it yet but thanks sooo much appletaartje for the simple explanation  :D

appletaartje, 2016-09-01:
@alan228 @aryangangurde 

I'm gald I could help, although it was such a simple thing...
I still tink that this should be somewhere a simple FAQ or something like that. But well, I think this also helps people to find out how things work.

If you guys have any other questions, just let me know. 
FYI im just built my very own first quadcopter (APM 2.8, Mission Planner, Turnigy TGY-i6 + TGY-iA6) and hoping to do some autotuning in the next couple of days (today seems there is to much wind...) 

Thanks and have a good day!

alan228, 2016-09-02:
I'm in the middle of building my first fpv tricopter but I keep breaking servos and my motors were cheap so I'm waiting on new motors in the mail. ..
I finally got it set up prefect and my early flying skills were not quite adequate to save it from a accelerated decent :-( it took me forever to figure out the oplink, mavlink osd, gps, PID's!¡!, and the 10ch update but finally it's all good and ready. ..for some motors, I built a few airboats waiting for the servo to come in cause I didn't trust the motors on my tricopter.  I'm running a hj-y3, revolution oplink fc, mavlink osd(5v mod), wr200 fpv tx, no name a2212 1000kv, no name 30a esc, op gps, flysky fs-i6 tx(10ch mod) with fs-ia6b rx on ppm with volt telemetry,  and more soldering than I ever thought I would have to do in my life lmao.  But seriously thanks again! I'll be ordering some fs-ia10b receivers after i order my new motors and esc setup

alan228, 2016-09-02:
More fun to be had though I ordered an led light, attopilot volt/current sensor so maybe I should start documenting on all the crap I've had to go through to get this far PID's being my least favorite thing to have to deal with but of course I'll be back at those again soon enough once I get my motors :-/ just wish some one had some clear help on the auto tune function it's really annoying going back and forth with the rate kp etc..

evyildirim, 2017-02-07:
hi guys,
i try for long hours to update, here what happens:
-i download .exe file (only .exe file... from [download] button) 
-run as admin
-open fw update mode(tried all=from hidden menu, from thr+yaw trim, from standard fw update menu)
-connect ftdi(tx+rx+gnd) pins
-open port (i can see current version)
- and here we go... when i click "programme" button, it stops at %1...or sometimes %3
loading of new fw never continues to 100% and i lost transmitter... 

i could load original firmware to open transmitter again. but cannot load 10ch fw.

any idea? any help?

thnx in advance