benb0jangles/FlySky-i6-Mod- #44: Disable RF Signal when using Simulators

Peter010103, 2016-11-02:
Hello,

Thanks for all the work you do! I would like to make a suggestion for the mod. firmware.

Could we have an option that disables the RF chip from outputting signal data when we use simulators? (i.e. have a 'Simulator Mode')... or is this not possible to do with a change in software? The signal interferes with wifi...

If it isn't possible, do you know any solutions to this problem?

MathijsG, 2016-11-10:
Duplicate of https://github.com/benb0jangles/FlySky-i6-Mod-/issues/33 but still having the same question. No reaction from anyone unfortunately. Would really come in handy, now its eating battery a lot while sim flying.

qba667, 2017-03-08:
@MathijsG, @Peter010103  
Work is in progress I will disable spi clock and we will see results.

shyam98, 2020-05-05:
Hello, 

was this issue ever resolved? 
Thanks :D