benb0jangles/FlySky-i6-Mod- #48: Altitude Sensor TGY-CAT01

DjFriday2000, 2016-11-17:
Hello,

It is possible to create a firmware to suport that sensor???
It works with TGY-i10 transmitter, with i6 doesn't.
I don't see this sensor in "sensor list". With i10 works fine i checked on my friend transmiter.It will be geat to use it on i6 radio.

benb0jangles, 2016-11-18:
There is discussion to change firmware to support telemetry by Ardupilot, cleanflight, and custom.
Ardupilot will have to use Mavlink to iBus convertor, cleanflight has ibus support, custom is ibus. All these have potential to have their telemetry data displayed, example: Current Flight Mode, GPS co-ordinates, Altitude, Distance from Home, Ground Speed, GPS Sat count, HDOP, Climb Rate, Throttle %.

Please ask here: https://www.rcgroups.com/forums/showthread.php?t=2486545&page=79

DjFriday2000, 2016-11-18:
Thanks, for your replay.
I will ask on rcgroups.

DjFriday2000, 2016-11-18:
I'm searching information about this sensor.
No one send a posts about it.
Still wondering why it works on i10 but on i6 doesn't.
I'm don't belive that producer use two differend protocols for reach one radios.I'm not programmer or electronic, my be rC groups resolve my problem.

madloki, 2016-11-25:
Hello, the latest changes in 10ch_Timer_MOD_i6_Programmer_V1_4 looks like nowadays transmitter will show all the sensors even if have no idea about how they work (V1_3), and there is even a short notice about ASL sensor namely (in V1_4). 

What exactly that means ? 

Does transmitter now display ASL sensor with proper name in V1_4 ?

Or even is this firmware able to count altitude from sensor data (has an proper algorythm) ?

Because that needs user input - at least set the starting (ground) position (ASL=0) of the plane somehow
from menu, to be able to show the proper altitude in meter/feet above groud from startpoint  ?

I'm not programmer, but I have FS-I6 and altitude sensor TGY-CAT01, so, If You want, I'm able to help with this, If You can.

For example, If just raw data are displayed now, I'm able to test this firmware at the ground on basement in our house, and then measure distance and write a raw data from staircase up to 4-th floor.

Maybe I can found a much higher public open building with possibility to measure first two or three floor altitude from staircase (then just multiplying shall be enough) ?

I suppose that altitude data from GPS are worthless for this purposes.

DjFriday2000, 2016-11-26:
Hello,

Madloki - i'm need help to start TGY-CAT01 on TGY-i6.Sensor is not apear in sensor list.I'm put 10ch_Timer_MOD_i6_V1_4 firmware and it doesn't.It loos like transmiter doesn't see this sensor.Would you help me with that??

greetings,

Djfriday

DjFriday2000, 2016-11-26:
Would you send me a photo how TGY-CAT01 looks like on I6 transmiter - at work! :)
Thanks for your answer.

Greetings,

Djfriday

jakubk, 2016-11-27:
@madloki - I am author of the TX mod - currently I am trying to run the sensor on I6 - but I have no sensor;). The sensor is different than other sensors - possibly special frame is send and then it is parsed in radio. Do you have maybe logic analyzer?

DjFriday2000, 2016-11-27:
Hello no i don't have.Wysłano z telefonu komórkowego Huawei

madloki, 2016-11-28:
sorry, but no logical analyzer there, just the tx and sensor

Tof68, 2017-03-15:
No news of this sensor? I would be very interested too by using it with i6 receiver...

madloki, 2017-03-15:
i'm giving up for now. Found plan&build a ~$20 vario using arduino + 443Mhz
rx/tx, with ability to switch vario/ASL on dedicated channel (asl beep for
houndreds and tens of meter using different tones). On banggood, just fs-i6
motherboards are available, no fs-i6x.

Fs-i10 is now for $100, because of more mixes, its better variant than
fs-i6x for me.

qba667, 2017-03-15:
@Tof68 basic implementation have been made. Some guys tested it and it seems to be working fine.

Tof68, 2017-03-15:
Sound good! does it mean that a 10ch_Timer_MOD_v1.5 going to be compiled and released?

cookieqk, 2017-07-27:
Hi, on the last qba667_hali mod there's just a pressure value - 9 or 10 digits like 345170... :-/
I can't get into =) extra =) ALT sens on fs-i6 menu.
It jumps to "tx voltage"

madloki, 2017-08-04:
Hello, the same with mine (10ch_MOD_i6_Programmer_qba667_hali9.exe
<https://github.com/benb0jangles/FlySky-i6-Mod-/blob/master/10ch%20qba667_hali9%20Updater/10ch_qba667_hali9_i6_Programmer_V1/10ch_MOD_i6_Programmer_qba667_hali9.exe>
).

I have arduino based vario meanwhile, but chance to use a FS-i6 only
solution one day would be fantastic. Arduino device + 430Mhz tx is a bit
heavier than CAT01, I need to have a 430MHz rx with speaker and battery in
my pocket. Altitude is not displayed, just speaker tones are used for, + 1
rx channel is occupied as vario/altimeter switch.

Now back to FS-I6 and TGY-CAT01 test:

These "numbers" are for sure pressurre (suppose in pascal). Each +1 meter
of altitude this number is decreased  (-12.5), and each -1m of altitude
this number is increased same way (+12.5)

That fits directly to altitude/airpressure condition in my country (central
europe) during normal summer conditions. I live ~650m above sea level.

Comparing to arduino vario code, this needs to be done in firmware:

ASL functionality:

You has to store initial value of the pressure in memory (start point = 0
meters), and compute positive or negative altitude comparing to start point
based on difference between stored and actual value divided by
approximatelly 12.5

This should give to us correct altitude. Think about negative values (slope
flying)

Vario:

vario mode (activated by switch) needs:

- take at least 4 values by 0.5 sec repeatedly into some "list"

If value is increasing for all values one by one, then compute difference
between first and last and divide by 25
You should got a vertical altitude change in m/s

If there is a speaker, this could be converted to some tone frequency based
on result value, if just buzzer is there, this could be represented on
display together with short and fast beep sequence for increase, or long
and slow beep sequence for dive.

For practical reson, just increasing altitude vario detection could be
implemented (not so loud and disturbing all the time).

best regards

qba667, 2017-08-04:
@madloki it is more complicated - but now I have the sensor so I can see what is send exactly.
There is barometric formula to calculate altitude. You can find it on my github and also in discussion on rcgroups. There is already variometer functionality in the radio - but it probably refers to distance...

madloki, 2017-08-04:
Hello, if You not insist on above sea level, then "above start level" is
such a simple. In europe up to 1000m above sea, no need to temperature and
humidity compensation. Just different temperature+humidity like
summer/winter switch is enough, by my experience. Travelling to mountains
over 2000-3000m make a difference, but not so much, if we talking about
relative (from startpoint, or during last 2 seconds) numbers.

qba667, 2017-08-04:
IT is logarithmic dependency and difference of temperature must be also considered.

madloki, 2017-08-04:
I don't care about precision You are talking about. For 400m range no
logarithm is needed. I don't care about exact value, just about its
relative changes.

Thats how most of cheap amateur barometer based ASL/vario works, including
common arduino ones. CAT-01 is cheapest, so no need for more precission
there.

qba667, 2017-12-11:
@DjFriday2000 @madloki @Tof68 
Finally implemented:
https://github.com/qba667/FlySkyI6/releases/tag/1.7.1

handstek, 2017-12-28:
Tried 1.7.1 but have problems with firmware so had to go back to 1.6 really want to use vario option

JensBoerner, 2018-02-16:
Hi,

Using IA6, IA6b and IA10 receivers i noticed currently (Firmware 1.75 for FS-I6) I only get an altitude in the sensor list for IA10. Actually I get Temperature and Pressure from CAT01 and an caclucated Altitude, but over Sea Level. In the Extra menu I can either change Pressure or temperature or set it to the current measures one, but not e.g. reset altitude to 0 to get Altitude over Ground.
In IA6B I only get Pressure, Temperature is always set to 25°, and I do not get any altitude or temperature in the sensor list.
I aim for updating the IA6B Formware, but not completly sure what version to choose and if this really is the problem.
Any thoughts on that? Is the problem an outdated IA6B Firmware having not all sensor data? Voltage, RPM and Temperature (extra Sensor) works as soon as I plug it in.

Oh and BTW. when addeding Temperature Sensor *and* CAT01 Pressure Sensor It does not work, the sensor list is shortened to end after the data received from the frist sensor, probably because CAT01 Temperature sensor data interfers with the FS-CTM01 Temperature sensor.