benb0jangles/FlySky-i6-Mod- #49: New timer Mod bug

Jahnkeanater, 2016-11-21:
The new timer will start randomly before the throttle is raised. Maybe make it stop counting when the throttle is lowered again or have it check that the throttle is raised for more then a split second. I think error in the throttle is triggering the timer.

Sometimes when the timer starts it will rapidly count up to 8 or 9 seconds.

Maybe a more convenient way to reset the timer from the main menu. The up down and cancel buttons don't do anything from the main screen so maybe a long press of the down to reset and a long press of the up to start/stop?

jakubk, 2016-11-27:
@Jahnkeanater can you try to set start condition to 1200?
Is the quick count triggered when leaving menu and start condition is true? 
I will add reset from main screen as well.

Jahnkeanater, 2016-12-06:
I have start set to 1500 and it still sometimes starts randomly. The quick count starts when raising the throttle past 1500. It only happens some of the time. I can't find a way to reproduce it. It must be when you raise the throttle at a certain point in the code. It only happens about 5% of the time.

Smacking the radio playing with switches and navigating the menus randomly I cannot get the timer to trigger.

Just letting the radio sit idle the timer starts. Typically 2 minutes after the screen goes dim. 

Does your radio not do this? Maybe I should re-flash the firmware.

Jahnkeanater, 2016-12-06:
Why are there 5 versions of the timer mod? Are you supposed to install both the 10 channel mod and the timer mod? Or is the timer mod supposed to be installed over original firmware?

jakubk, 2016-12-06:
@Jahnkeanater  I can imagine that there is some period causing this strange quick counting. I had to add double time compensation because using calculated tick resolution radio seems to be slowed down almost twice when calculating difference. It looks like they are number of interrupts slowing down main loop - main loop is controlling timer. I will try to add own interrupt for counting time - this should solve the issue.
I will observe if timer starts automatic after some idle time - but it looks like you are getting invalid values for throttle. Does it happen when you assign SWA or other switch?
Timer and telemetry mod is using as a base latest 10 channel mod. So just install the latest.

Jahnkeanater, 2016-12-07:
I normally have all the switches assigned to channels. On a fresh model with default aux channel assignment the timer still starts.

I reset the flash memory and re flashed to original and then the latest mod and I am still getting the same behavior.

Jahnkeanater, 2016-12-08:
Do you have asm files for the i6? I have experience with the Motorola hcs12. Are the op codes the same for the freescale chip?

jakubk, 2016-12-08:
The assembly language used for the MCU is ARM thumb - let say it is similar to hcs12. I will attach disassembly today evening.
If you have some are debug tool I can tell you where to set breakpoint - otherwise we can add displaying of start conditions on main screen.

jakubk, 2016-12-09:
@Jahnkeanater  As promised
[tx_full_backup_2.zip](https://github.com/benb0jangles/FlySky-i6-Mod-/files/642715/tx_full_backup_2.zip)

qba667, 2017-01-24:
@Jahnkeanater I can reproduce the issue lately - I will check what is wrong.

qba667, 2017-03-22:
@Jahnkeanater solved in last version published on rcgroups.