benb0jangles/FlySky-i6-Mod- #50: won't stay bound?

reticentnz, 2016-12-10:
I did the 10 channel mod and everything works great. One thing I have noticed which is a bit of a pain is that it won't stay bound. When I power up the TX, I need to hold the bind key every time my quad has been restarted. 

ie, power quad, ia6b stays flashing. Turn on TX with bind key held. Binds OK. I can then turn reciever on/off as much as I like while quad stays powered. When I reset the quad power I need to begin the process again. 

Only started happening after the mod - is this an expected change in behaviour? Any ideas? Cheers

johnryanlee, 2016-12-11:
I have not experienced this myself.  Maybe reset your TX to factory
settings (not firmware) and re-bind.  See if that helps.

reticentnz, 2016-12-12:
Same thing unfortunately, even after the reset. Is there anyway to 'reset' the receiver at all? I've had a bit of a look around at couldn't see what the 'update' button is used for.

Jahnkeanater, 2016-12-14:
I doubt this it a problem with the radio. Once bound a receiver will still connect to any model memory. There is no unique id for each model. Maybe your receiver is going into bind mode? Make sure nothing is shorting out the bind pin inside the receiver.

I believe you can update the firmware on the receiver but I don't think there is really any reason to do this. I don't think it adds any functionality.