benb0jangles/FlySky-i6-Mod- #51: Timer Control

Peter010103, 2016-12-16:
Thanks for making such awesome firmwares! 

I recently installed the timer mod firmware but here are some suggestions to improve the current version. Although the timer function is great, I wish there was an option to make the buzzer beep when it is over a certain time (which the user can set).

qba667, 2017-01-24:
@Peter010103 
I will try add it but first I have to rework storing settings in eeprom. To get extra space for our additional settings. Anyhow on my todo list.

PopPopFPV, 2017-01-31:
I too updated to the timer mod an am pleased with it.
You're probably already coding for it, but it would be nice to be able to stop/reset the timer without having to reset the transmitter. Any plans for this?
Keep up the great work.

qba667, 2017-03-08:
@Peter010103  future almost finished - I will keep you updated.