benb0jangles/FlySky-i6-Mod- #52: FS i6s support

jacobbloy, 2016-12-17:
Is there any way to get this working for the fs i6s I would love to do some changes to the firmware

qba667, 2017-03-08:
@jacobbloy as long as we have no source and mcu is different nothing can be done - I can give you my disassembly.. but way to make something working is very long.

jacobbloy, 2017-03-08:
Any thing you can provide will help. We have the i6s updated that seems very similar to the the i6 updater. But as you said the mcu is different.

qba667, 2017-03-08:
I believe code base is the same as well. Check here - https://github.com/qba667/FlySkyI6
I had to remove IDA project because of license but I can send you it.

jacobbloy, 2017-03-08:
That would be great. My email is jacob.bloy@gmail.com

message2prateek, 2019-02-07:
Will love you see support for fs-i6s. I think it's a really good radio but is crippled by its firmware.

qba667, 2019-02-07:
@message2prateek, @jacobbloy  I just got I6s from banggood - as samples for development. Maybe I will be able  to implement something on it - I am working on OpenTx port for M0 architecture.

message2prateek, 2019-02-07:
Awesome! OpenTx on FS i6S will make it the best value radio.

ncornejo, 2019-10-31:
> @message2prateek, @jacobbloy I just got I6s from banggood - as samples for development. Maybe I will be able to implement something on it - I am working on OpenTx port for M0 architecture.

Some news about this ?