benb0jangles/FlySky-i6-Mod- #53: Extra channels on trainer port?

rabno1, 2017-01-01:
Is there any way to get the extra channels to work through the trainer port ppm output?  Love the longer timer and the ability to mix 2 switches on 1 channel (6 flight modes). But as I don't use a RX, I use a oplink (inside the TX hardwired to the back of the trainer port)  I still only get 6 channels. Is this just a setting I'm missing or is it an actual issue with the firmware?

qba667, 2017-01-24:
@rabno1 It is on my to do list.

rabno1, 2017-01-24:
Cool look forward to it being implemented.

qba667, 2017-02-15:
@rabno1
check my last change on RC groups. Now 8 channels are send out.
https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch%21/page96