benb0jangles/FlySky-i6-Mod- #54: Programmer won't connect to FlySky FS-I6

Eric032990, 2017-01-04:
Im running wine bottler on a Mac. Im connecting the USB cable to trainer port on the transmitter to COM4 and when I click open in the programmer it opens the port but the transmitter never actually connects, I tried COM3 and the same thing. Any idea whats going on or what Im doing wrong??

lrusnac, 2017-01-04:
I used this skript for updating the firmware on mac (the hex from this repo and the go script from the linked one) https://github.com/mhils/flysky-updater

hope this will help

Eric032990, 2017-01-05:
Not entirely sure on how to use that script....

keir-rex, 2017-01-18:
@Eric032990 I had difficulty getting this working on both my Mac and Windows 7 running in a VM on my Mac. I tried using the script suggested above to no avail. It worked without issue on a Windows 7 computer so I'd recommend you save yourself the hassle and borrow someone's Windows computer.

LazaroFilm, 2017-03-09:
Funny thing, but I had an issue too and tried swapping the tx and rx wires in the plug and it worked...

Dragonqos, 2017-07-01:
How to update firmware using MacOS.

1. First you need USB-UART PL2303HX (or a cable if you have, or any other USB-UART)
2. Plug in and go to http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41 to get drivers.
3. Install driver, after finish reboot your computer
4. Download HEX file "flyplus_beta_20_01_16.hex" from current repo and put it to the folder ~/Flysky
5. Download files "build.py" and "flysky_updater.go" from https://github.com/mhils/flysky-updater and put them again to the same folder  ~/Flysky
6. Install Golang with dependencies:
    > cd ~/Flysky
    > brew install go
    > go get github.com/cheggaaa/pb
    > go get github.com/tarm/serial
    > go get gopkg.in/alecthomas/kingpin.v2
7. Connect USB-UART the same way as it was described README.md and run firmware update mode.
8. Run updater with command:
    > go run flysky_updater.go /dev/tty.usbserial flyplus_beta_20_01_16.hex

Thats it.﻿

olivCode, 2018-02-02:
I have an issue with @Dragonqos method on my mac.
First I had to add a dependency : go get github.com/AlecAivazis/survey
After that I could launch the update but encountered another error : 
flysky_updater: error: unexpected /dev/tty.usbserial, try --help
exit status 1

Any idea ?

vchoubard, 2018-02-16:
@olivCode run:
`go get github.com/AlecAivazis/survey/core`

Jebasteen, 2018-03-14:
can i upgrade fsi6x to 10channel ?because it shows content only for i6 in youtube 
and i having flysky data cable .itS not detecting my transmitter help me iam in urgent

amckee, 2018-07-12:
As stated by @LazaroFilm I also swapped tx and rx wires with success.

HilltopFPV, 2019-04-19:
i'm trying to do this on windows 10, i'm using an arduino like people say to in videos, they say to jump the reset pin and gnd, my computer cannot detect the arduino when they are briged any suggestion?
thanks