benb0jangles/FlySky-i6-Mod- #55: Trainer and student mode

pigiit, 2017-01-05:
Hi,
I upgraded my fs-i6 and it works very well, great work!
..but maybe I found an issue, not sure if related to your upgrade or original firmware, using student option.

when student option is active the radio transmits both RF and PPM signals on all channels, in my opinion should mute the RF (it's supposed to be connected to the master that is the only one to go onair).

Moreover could be nice to have, when in Master mode, also the PPM out also for CH1-2-3-4 that at the moment is not transmitted anymore, with this option should be easy to add an head-tracked device, putting in master, reading the PPM out, adding the extra channel (7-8 for example) and injecting all back to the radio as PPM IN. As far as I know that's not possible now.

Thanks

qba667, 2017-03-08:
@ working I am working on the issue and future request.

usonic777, 2020-01-25:
Hello,
I was also looking for option to disable RF transmitter and keep only PPM. Was that implemented on some version of firmware?
And thank you for your work which is done already and works great!