benb0jangles/FlySky-i6-Mod- #56: Flashing Eachine i6 transmitter (rebranded FlySky I6)

alfag4, 2017-01-06:
Hello, is it possible to flash the Eachine i6 Transmitter to original FlySky i6 firmware and then flashing it with 10ch firmware?
I have also X6B receiver in iBus "mode".

pigiit, 2017-01-06:
I flashed my Eachine i6 Transmitter and works fine, i use it with rx iA6B

alfag4, 2017-01-06:
Thanks,
i have an usb-> serial converter based on FTDI FT232RL. Can i use it to flash via the trainer port?

pigiit, 2017-01-07:
I used a serial converter similar to the one you have that works at 5v and not at 3.3v, but I had no issues.
.

alfag4, 2017-01-07:
Thanks a lot

Woobatron, 2017-02-22:
I can confirm the CH340 type usb-to-serial adapter as well.

crozone, 2017-02-22:
I used a homemade cable with an FTDI adapter and it flashed fine. I use it with an FS-iA10B receiver and get all channels over PWM without issue.

yo8aiv, 2017-05-31:
used a rs232 usb to ttl D-SUN. flashed with exe ver1.5 / flyplus_beta_20_01_16.hex, and it works with ibus on stock receiver FS-I6X - eachine V-tail 210.
Thinking on trying voltage telemetry now.

big thanks. Great work.