benb0jangles/FlySky-i6-Mod- #57: 10 CH doesnt work in AFHDS mode (not 2A)

piodabro, 2017-01-08:
Is transmiter should push 10 ch in AFHDS mode?
Now when i try to use with qx95, there is only 6ch present. In 2A mode 10ch works good.

zelwake, 2017-02-20:
thats receiver issue....most AFHDS receivers are only 6 channel, a8s has only 8 channels over ppm

qba667, 2017-03-22:
@piodabro the response from @zelwake is correct all 14 channels are transferred over radio link - modes are not distinguished while creating radio packet.

qba667, 2017-04-14:
@piodabro Small update there is difference in AFHDS. Here you have changed FW but not tested.
[10_ch_AFHDS_updater_04_14_22_36.zip](https://github.com/benb0jangles/FlySky-i6-Mod-/files/922918/10_ch_AFHDS_updater_04_14_22_36.zip)

piodabro, 2017-04-14:
@qba667 i asked about AFHDS (not 2A) cause my QX95 (integrated reciver) recives only 6ch. I thought that is problem with AFHDS but maybe its reciver's fault. I do not worry about that anymore, just use only 6ch.
What changes this update?

qba667, 2017-04-20:
Now 10 channels are transmitted over the air. Other changes as described in https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch!