benb0jangles/FlySky-i6-Mod- #58: Subtrim on Aux channels 7-10 & Source code

keir-rex, 2017-01-19:
Hey @benb0jangles, 

I'm wondering if there's a limitation preventing implementing subtrim for Aux channels 7-10?

If there's no limitation - would it be possible for you to provide the source from which you compiled and development tools/environment/methods? I'd like to get started hacking on the firmware and I'm sure I'm not the only person who would benefit from the subtrim on these channels.

Thanks,

Keir

keir-rex, 2017-01-19:
Feel free to give me a slap in the face if I'm missing something obvious - I'm hoping you're not editing the hex files directly... 😬

qba667, 2017-01-24:
Let me know if you are still interested. Some help will be appropriated. I was never able to compile whole firmware from disassembly but I was working with chunks. Bacillary assembler for ARM M0+ compiled with GCC. And some python script to calculate checksum.

keir-rex, 2017-01-25:
Yeah, I'm still interested. Do you have a repo for tracking the firmware development?

qba667, 2017-01-26:
I have no repo right now, possibly I will try to create it during weekend.
Here you have method of your interest. As you can see most of settings are affecting 6ch only.
To change it you have redefine model settings array to be able to store all new configuration.
```c
int *createRadioPacket()
{
  unsigned int v0; // r0@2
  int v1; // r0@5
  int *v2; // r1@5
  signed int v3; // r2@5
  unsigned int ch_index; // r4@11
  unsigned int v5; // r7@19
  __int16 *v6; // r5@19
  int v7; // r0@19
  unsigned int v8; // r1@20
  int v9; // r0@25
  unsigned int v10; // r0@25
  int v11; // r0@27
  int v12; // r1@27
  int v13; // r0@32
  int v14; // r0@38
  int v15; // r0@41
  int v16; // r0@48
  int v17; // r0@55
  int v18; // r0@57
  int *tmpCH_1; // r0@59
  int *tmpCH2; // r0@63
  int *tmpCH3; // r0@67
  int *tmpCH4; // r0@71
  int *tmpCH5; // r0@75
  int *tmpCH6; // r0@80
  int ch_index2; // r5@86
  int v26; // r0@90
  int v27; // r7@90
  int chValue; // r0@90
  int v29; // r0@92
  unsigned int v30; // r2@92
  unsigned int v31; // r0@92
  int v32; // r12@92
  unsigned int v33; // r0@99
  int v34; // r0@102
  unsigned int v35; // r0@103
  int v36; // r0@105
  int v37; // r7@112
  int v38; // r6@112
  int v39; // r0@112
  int v40; // r1@112
  int v41; // r2@114
  int v42; // r1@114
  int v43; // r0@117
  signed int v44; // r3@120
  int v45; // r5@123
  unsigned int v46; // r6@123
  int v47; // r1@126
  int v48; // r0@126
  int v49; // r1@127
  int v50; // r2@127
  int v51; // r6@133
  int v52; // r4@134
  unsigned int v53; // r3@134
  unsigned int v54; // r0@135
  int v55; // r1@136
  unsigned int v56; // r7@138
  int v57; // r0@139
  unsigned int v58; // r0@146
  int v59; // r5@149
  int v60; // r0@149
  int v61; // r5@151
  int v62; // r0@151
  unsigned int v63; // r4@153
[... 489 lines cut ...]
  ch_4 = v75;
  dword_1FFFFDEC = v75;
  v76 = (signed int)ch_5;
  if ( (signed int)ch_5 <= (signed int)0xFFFFD120 )
    v76 = 0xFFFFD120;
  ch_5 = (int *)v76;
  if ( v76 > 0x2EE0 )
    v76 = 0x2EE0;
  ch_5 = (int *)v76;
  dword_1FFFFDF0 = v76;
  if ( ch_6 > (signed int)0xFFFFD120 )
    v68 = ch_6;
  ch_6 = v68;
  if ( v68 <= 0x2EE0 )
    v71 = v68;
  dword_1FFFFDF4 = v71;
  v77 = currModelNamePtr;
  dword_1FFFFDF8 = (int)createPacketsForCH7_10(*(_BYTE *)(currModelNamePtr + 141) & 0xF);
  dword_1FFFFDFC = (int)createPacketsForCH7_10((unsigned int)*(_BYTE *)(v77 + 141) >> 4);
  dword_1FFFFE00 = (int)createPacketsForCH7_10(*(_BYTE *)(v77 + 145) & 0xF);
  result = createPacketsForCH7_10((unsigned int)*(_BYTE *)(v77 + 145) >> 4);
  dword_1FFFFE04 = (int)result;
  return result;
}
```

keir-rex, 2017-01-27:
@qba667 thanks for that - I can create a repo and invite you as an admin. I'll have some time over this weekend - I'm GMT what time zone are you in?

qba667, 2017-01-27:
@keir-rex sounds good, have you any experience with disassembly? I am in GMT+1.

keir-rex, 2017-01-27:
@qba667 not really - this is going to be a steep learning curve. It'll be nice to see if we can apply some proper engineering principles and improve the agility and pace.

Anything in the way resources, literature, pointers, and existing dev-environment you can provide would be appreciated!

qba667, 2017-01-27:
@keir-rex same as it was mine. 
To do anything you will need debugger. I am using J-Link but STLink will be also fine.
Toolchain is part of Kinetis Design Studio Integrated Development Environment.
SDK for MKL16z (http://www.nxp.com/jp/products/software-and-tools/run-time-software/kinetis-software-and-tools/development-platforms-with-mbed/software-development-kit-for-kinetis-mcus:KINETIS-SDK)

Some resources about MCU
http://javaarm.com/file/ARM/books/Cortex/The.Definitive.Guide.to.the.ARM.Cortex-M0_Joseph.Yiu_2011.pdf

ARM thumb 
http://www.keil.com/support/man/docs/armasm/armasm_dom1359731139853.htm
http://www.embedded.com/electronics-blogs/beginner-s-corner/4024632/Introduction-to-ARM-thumb
https://ece.uwaterloo.ca/~ece222/ARM/ARM7-TDMI-manual-pt3.pdf

Notepad++ 

And the most important part a lot of free time.

qba667, 2017-01-28:
@keir-rex check here:
https://github.com/qba667/FlySkyI6

Tof68, 2017-03-15:
Subtrim and end-points for channels 7 to 10 should be realy great. It's almost necessary when using this channels with servos, even to drive accessories

Pecacheu, 2017-04-02:
@qba667 I can't figure out how to compile the source code. I made an issue about it [here](https://github.com/qba667/FlySkyI6/issues/2).

keir-rex, 2017-07-04:
Apologies for not getting in touch for circa 5 months - I've not looked at my RC stuff since the weather here just isn't suited to it. I'm going to try put some time over the coming weeks.

kelvinpmr, 2017-08-28:
@keir-rex hi man! Please! Do the endpoint for this ch7! ALL naza users Will be thx to you!