benb0jangles/FlySky-i6-Mod- #59: New sensor type

hali9, 2017-02-05:
Hello.
At the beginning, I would like to thank you for what You do.
In 10ch_Timer_MOD_i6_Programmer_V1_4 You add alt sensor support 
(id:0xf9, display: ALT, unit: m, type like RPM).
Can You add more sensor type/id as bellow ?
ID: 0x20 
Display: STAT
Type: like RPM
For display status, no unit (or space).

ID: 0x21
Display: CURR
Unit: A
Type: like voltage
For handle current from flight controler.

ID: 0x22 
Display: ALTB
Unit: m
Type: like voltage
For handle baro/sonar signed/unsigned altitiude, for example -1,23m or 123,45m

ID: 0x23 
Display: COUR
Unit: °
Type: like RPM
Course (Mag, GPS, other) in degree (if ° is not passible then *).

ID: 0x24 
Display: DIR
Unit: °
Type: like RPM
Direction to home in degree (if ° is not passible then *), the same as COUR but display DIR.

ID: 0x25 
Display: DIST
Unit: m
Type: like RPM
Distance to home, the same as ALT but display DIST.

qba667, 2017-02-15:
@hali9 
The changed to sensor code was made by me. I will rewrite this code soon. Meantime please check ids used currently some of them are matching your names but I was using different ids.
Please check this file:
https://github.com/qba667/MAVLinkToIbus/blob/master/MAVLinkToIBUS/IBUSTelemetry.h

hali9, 2017-02-15:
Hello @qba667.
Thank for respond.

This post refers to 10ch_Timer_MOD_i6_Programmer_V1_4 and this soft support only a few additional useful ids (like alt).

Currently I use 10ch_Mavlink_MOD_i6_Programmer_V1. This solutions is very good and almost perfect, but almost:

There are more ids sensors and they are it own name, but not have a unit and they are not formating correctly.

For example:
GAlt have this same ids as overwrited ALT, it is in m (meter) but 1m=100 and it is formatting like RPM.
I think standard ALT is better because can support 65535m and GAlt only 655m.
Should be standard ALT (or change only label to GAlt) for GPS altitude.
And should be a socondary ALT for baro as I write above (formatting like voltage and with unit m -1,23m or 123,45m).

The Current sensor should have unit A (amper).

Heading is formating like voltege (100.00). 
Should be like RPM (100) and have unit ° or * or deg.
I propose add secondary heading for direction to home and with label DIR or  HomeDir or DIRH or something like this.
And third for GPS course (like COG but formating like RPM with ° unit) and Heading use for magnetic sensor.

Should be a main status sensor (STAT) with no unit.
Armed, FlyMode and GPS is ok but it is 3 sensors, better pack more information in one sensor, because FlyskyI6 can receive only 15 sensors. Of cours may support difrents ids.

So I think my post is current for 10ch_Mavlink_MOD_i6_Programmer_V1 too.

I propose also sensors type for GPS coordinate.
LON1 as ExtV -12.45 (-12.3456789 N)
LON2 as RPM 5678 N/S
LAT1 as ExtV -123.45 (-123.4567890 E) 
LAT2 as RPM 6789 E/W

One of coordinates requires 4 bytes. You can just smash it in the two sensors with two byte each, 
respectively format and display. Unless you could combine them in i6 and view it as one.

If you would add to this 6 sensors on the main screen, together with the current modifications and updater would be really ideal.

Thank you for your attention.

qba667, 2017-02-16:
@hali9 good to know that someone is using my implementation.
I was aware about most of issues but not all of them. Currently I am working on rewriting the code in C while my old asm implementation is hard to extend and maintain. The workaround for GPS sensors will be AC frame implementation so we can have 15 sensors with 28 bytes each - we can put a lot of data there. 
Number of sensors limited to 15 is just RX limitation, in Tx there is 256 bytes length array reserved for sensors data.
Have you tried also my experimental 6 sensor on main screen version from RC groups?
Currently I have also changed format of SNR/Voltage/Timer they are displayed as first line, then space for 6 configurable sensors.  I am also working on multiple pages on main screen so we can configure more sensors.

hali9, 2017-02-16:
I didn't try skip_AD_request.bin. I cannot flash it.
I can test it, but I need updater like 10ch_Mavlink_MOD_i6_Programmer_V1.exe, because I have Flysky data cable (http://www.banggood.com/FlySky-Data-Cable-USB-Download-Line-For-FS-i6-FS-T6-Transmitter-Firmware-Update-p-982289.html?rmmds=search).
Thom updater not work for me (application crash when "Detecticting ...").

No, I not using your sensor emulator. Better is implemented it in code in FC.
I use inav, so I create PR in inav https://github.com/iNavFlight/inav/pull/1228.
This work well, but problem is with formating. 
Two bytes for sensor is enough (exception GPS).
The sensors also can not be too much because on the screen, you can fit up to 6 max and 2 in icon (Tx.V Ex.V).

AC frame seems to be a noble idea, but whether it needs to upgrade the receiver?
I use stock FS-Ia6B and for flash it I need ST-Link. I dont have any, but is this will be good?
http://www.banggood.com/STM8-STM32-Emulator-Simulator-Programmer-Downloader-For-ST-LINK-V2-With-4Pin-Cable-p-1080420.html?rmmds=search

In my opinion better is properly handle several sensors and it would work really great.
Especially that these modifications are few, and they work really well.

If these modifications do updater like 10ch_Mavlink_MOD_i6_Programmer_V1, this will be a simple and very useful for many people, and to flash, it will required only Flysky cable or Ardunio and no need RX upgrade and STlink.

qba667, 2017-02-16:
@hali9 I will check provided repository - but looks good so far!
Updater is being created by benb0jangles so I can not help here.
In Thom's updater try to specify port and switch off auto detection.
ST Link is just fine but search at aliexpress it should valued at about 2$.
AD frames are strange - they were send when I was using long responses - maybe it is issue with my emulator - invalid timing or something but my responses overlaps some memory area and because of that AD frames were generated.. It must be checked.

hali9, 2017-02-17:
I managed to upload multiSensor_updater_01_11_18_21.bin using ThomHPL/BaseI6_updater by Flysky data cable.
This not work on WinXP, and not wor on Win7.
Open port succesed, but when "Detecting ..." the TX application crash.
 
Only work on Win8.1Pro.
I will test it.

What are the exact differences between the multiSensor_updater_01_11_18_21.bin and skip_AD_request.bin.

qba667, 2017-02-17:
@hali9 skip_AD_request.bin is changed firmware for IA6B (when long telemetry data was used it keep sending AD frames so I disable checking) - this is not for flashing TX.
The multiSensor_updater_01_11_18_21.bin will allow you to configure 6 sensors on main screen - other issues are not resolved.

hali9, 2017-02-22:
I tested multiSensor_updater_01_11_18_21.bin, it is work well.
But font on main screen are too small (cannot read when fly), should be standard font.

qba667, 2017-02-23:
@hali9  I have one more version where I have used more readable font 5x7 - I can publish this file. Have you notices anything strange with this version? How about formatting? I have written this code in assembly and it is hard to change. I was working on c variant but I have some trouble with linker It wont put the code at address where I need it.

hali9, 2017-02-23:
Publish wersion 5x7 if You can. I can test it.
I did not notice any strange behavior.
Formating is as I write above, repetition to organize:
Format:
Current sensor should have unit A (amper).
Heading is formating like voltage (123.00), should be like RPM (123), and should have unit ° (degree).
COG is formating like voltage (123.00) , should be RPM (123), should have unit ° (degree).
GAlt is in m (meter) but 100=1m, sholud be 1=1m.
New:
Add Status (only change label for s85).
Add Dir. (only change label for s86).
Add ALT for baro formatting like voltage -1,23m or 123,45m and with unit m (can use s87).
Add LAT1 formating like voltage -12.45 with no unit (can use s88).
Add LON1formating like voltage -123.45 with no unit (can use s89).

qba667, 2017-02-23:
@hali9 The formatting of lat lon is more complicated but will be done.
FW used by me - other font. If you create some nicer font - eg 6x8 or 7x8 I can use it. The oryginal 8x8 const width is useless.
[_updater_02_19_11_20.zip](https://github.com/benb0jangles/FlySky-i6-Mod-/files/797280/_updater_02_19_11_20.zip)

qba667, 2017-02-26:
@hali9 is there any reason to send LAT and LON twice? You have defined LAT1/LAT2 and LON1/LON2.

hali9, 2017-02-26:
Yes @qba667 , because for representing LON or LAT is need 4 byte. 
And Flysky telemetry supports only 2 byte value.
This is enough for all sensor but not for LON, LAT.
So, I split LON to two sensor LON1 and LON2 for 2 bytes each.
And LAT to LAT1 and LAT2.

Now I have for example for LAT (-12.3456789)  LON (-123.4567890)
LAT1 as ExtV -12.45
LAT2 as RPM 5678
LON1 as ExtV -123.45 
LON2 as RPM 6789
This can be send in 4 sensors for any Flysky RX/TX, stocked too.
This need formatting as I write before.
LAT2 LON2 is ok for sensor id 0x82, 0x83.
But there are not LAT1 LON1.

qba667, 2017-02-27:
@hali9 "And Flysky telemetry supports only 2 byte value." That is not true you can send 4 bytes values as well - without any modification. Just check here:
https://github.com/cleanflight/cleanflight/issues/1125#issuecomment-260173501
4 byte values are working in my preview version it will be published soon.

hali9, 2017-02-28:
@qba667 Can You publish version from this photo ?
https://github.com/cleanflight/cleanflight/issues/1125#issuecomment-269458664

qba667, 2017-03-04:
@hali9 I don't know where I have it. But soon I will publish completed version and source code for changes.