benb0jangles/FlySky-i6-Mod- #60: Loading Stops at 1%

evyildirim, 2017-02-09:
hi guys,
i try for long hours to update, here what happens:
-i download .exe file (only .exe file... from [download] button)
-run as admin
-open fw update mode(tried all=from hidden menu, from thr+yaw trim, from standard fw update menu)
-connect 3.3V ftdi(tx+rx+gnd) pins
-open port (i can see current version)

and here we go... when i click "programme" button, it stops at %1...or sometimes %3
loading of new fw never continues to 100% and i lost transmitter...
i could load original firmware to open transmitter again. but cannot load 10ch fw.

any idea? any help?

edit: i m using windows10

thnx in advance

qba667, 2017-02-15:
@evyildirim 
Are you using FlySky branded transmitter?

evyildirim, 2017-02-15:
@qba667 
yes, FLYSKY,
i tried with another pc(win8), nothing s changed.

what i noticed during this update:
-when i run .exe file as admin => "MSCOMM32.OCX" file comes in main folder(where .exe file is located)
-when i click "programme" button => "demo.anp" file comes in main folder...

is this normal? or is it related with a problem?

regards

qba667, 2017-02-15:
That is strange, have you tried ThomHPL updater?
https://github.com/ThomHPL/BaseI6_updater
Is that brand new unit? Maybe something has changed in FW.
What is production date?
It that will not help possibly you have to use ST-Link.

evyildirim, 2017-02-16:
@qba667 
i will try it. 
BTW, when it stops at 1%, i terminate operation.
so, i cannot open tx again but i can connect to pc and load original FW(25 Nov 2015). this is the only FW that i can load.

evyildirim, 2017-03-21:
@qba667 
hi,
yes! it worked :) thank you very much. i updated tx in a minute.

regards