benb0jangles/FlySky-i6-Mod- #64: Channel 6 locked to VRB (V1.3/1.5)

Mattie19, 2017-02-24:
In V1.5 I noticed that channel 6 in variable pitch helicopter mode is not mapped to the throttle stick, but instead it is locked to VRB. When I go into AUX channels, it says channel 6 used and cant be changed. Im forced to go back to V1.2 since this issue is not there. Controlling my pitch with the VRB knob is a bit weird:D Could this issue please be fixed? 

btw, is V1.2 stable enough to fly?

qba667, 2017-02-26:
@Mattie19 - just reproduced this one. So value of channel 6 should be mapped to throttle stick - any curves are used?

Mattie19, 2017-02-26:
@qba667 Correct! Yes I do need to be able to use expo. I could mix channel 3 with channel 6 and leave VRB at the 0 position, but then if I use throttle curves, the pitch channel will be messed up. Basically the channel mapping is wrong compared to the original firmware, I hope this can be fixed.

Thanks!

qba667, 2017-02-26:
@Mattie19 - I remember that once Dave (author of original mod) has changed something - because also there were problems with pitch or something. There is some compassion of model type and different values are assigned to channel 6. I can try fix this one but I need to understand the concept. Can you provide me some resources? Or maybe just describe how it is working. I seen in few places in assembly that model type is compared and then some other values are assigned to channel 5 and 6.

Mattie19, 2017-02-26:
@qba667 I will do my best to explain this. When you select the Variable Pitch Helicopter type, Channel 3 and 6 should be connected to the throttle stick (in mode 2 this is the left vertical stick). So they are both connected to one physical stick axis, but they are seperated as a channel (3 and 6). In the menu you will find "Throttle Curve" & "Pitch Curve". The trottle curve will only affect channel 3, and pitch curve will only affect channel 6. So when we have linear curves, when the throttle stick is down, both channels are -100, and with full throttle stick, both channels should be at 100.

If you dont know collective pitch helicopters, they work like this: When you throttle up, the motor should spin faster, but the collective pitch of the helicopter blades should also change to get more air bite. If you flash the FS-i6 to the original firmware, or v1.2, you will see exactly what it does.

Thanks for the help, really appriciate it!

qba667, 2017-02-26:
@Mattie19 Good explanation - I got the point. Curves definitions must be found and calculation must be performed, the code should be there so we have to find why is not executed. The disassembly of original firmware must be also checked. I will let you know!

Mattie19, 2017-02-26:
@qba667 Yes, I would assume that this was accidently changed along the developement. Im not developing for this device, but in theory I think this should be an easy fix. The channel is just assigned to the wrong stick/switch/knob. 

I hope you are able to fix this issue, would be awesome to use 1.5:)

qba667, 2017-02-26:
@Mattie19 I found the place:
```c
if ( modelType_ < 3 )
  {
    createPacketsForCH6_10(*(_BYTE *)(modelPtr3 + 61));
    endpoints[7] = v51;
    ch_6 = v51;
  }
```

the method createPacketsForCH6_10 was defined by Dave model type in your case is 2 so it is over simplified - because always using channel configured in aux channels so in default case VAR B.

qba667, 2017-02-26:
@Mattie19 - bingo - one byte change:)
[heli_var_pitch.zip](https://github.com/benb0jangles/FlySky-i6-Mod-/files/802151/heli_var_pitch.zip)

Mattie19, 2017-02-26:
@qba667 Wow that was quick! Really glad you fixed it in code. 
Just one more question, how can I use the bin files to flash the i6?

qba667, 2017-02-26:
@Mattie19 I am using this one:
https://github.com/ThomHPL/BaseI6_updater
just place the file with updater in name in folder where Run.bat is. Then change in Run.bat in line
bin\1.2\baseI6_updater -a -u -p path to firmware image file here -c -v
"path to firmware image file here" to file name 
start bat and it should be fleshed.

Mattie19, 2017-02-26:
@qba667 Thank you for the explenation, it worked!!! 
I wasn't really sure what heli_full_02_26_19_10.bin was doing, but the updater file did the trick.
Maybe it's a good idea add these files with instructions so other helicopter pilots can have this fix too? https://github.com/benb0jangles/FlySky-i6-Mod-/issues/32 As you can see this issue was mentioned before, but it never got fixed.

Again thank you so much, you really made my week!

qba667, 2017-02-26:
@Mattie19 you are right - same issue - but I have no rights here. Write to benb0jangles he can pack file into updater and provide description.

Mattie19, 2017-02-26:
@qba667 Will do, I was super confused when I flashed my i6 and CH6 wasn't working properly. 
After flashing multiple versions and testing all types, I knew that it wasnt supposed to be like this. Luckely you were able to fix it within a day, good stuff!

Mattie19, 2017-02-27:
@qba667 I found one little bug. In the "End Points"  settings, channel 6 is still mapped to VRB. 
How to test: select channel 6, move throttle stick up and down, nothing happens. When turning the VRB knob, arrow does move from left to right.

qba667, 2017-03-22:
@Mattie19 in free time I will try address the issue.

JX5S, 2018-06-11:
Hi,
would it be possible to somehow compile firmware using channel 5 instead of channel 6? Or is there any way to do it currently? I need to have throttle mirrored to channel 5 and see no other choice than a firmware modification and I don't understand this at all...
Thanks in advance, Jakub

qba667, 2018-06-12:
@JX5S 
I think it can be done - just instead using PPM values channel values must be used - small change. 
Without firmware modification you can connect PPM out to PPM in and then select PPM 3 as source for channel 5.

JX5S, 2018-06-12:
Oh, I had no idea what ppm in aux channels meant. After looking online what its for this sounds doable. I'll try it when I get a chance. Thanks for your suggestion, now I understand my transmitter a tiny bit more 😄