benb0jangles/FlySky-i6-Mod- #67: Simulator / Student mode channel 5 and channel 6 hard coded

spuder, 2017-03-26:
I installed the 10 channel firmware and it is great. 

However in simulators (specifically the DRL simulator), I can no longer set the reset quad / reset race switches to SWB and SWC like they are on a stock i6. 

It seems that channel 5 and 6 are hard coded to use VRA and VRB when in student mode.

qba667, 2017-04-06:
@spudler How about trainer mode?

spuder, 2017-04-10:
Never used trainer mode. I can only test student mode.

spuder, 2017-04-24:
The readme in the stock firmware mentions that channel 5 & 6 are hard coded to VRA and VRB

https://github.com/benb0jangles/FlySky-i6-Mod-/blob/master/Firmwares/Flysky%20i6%20Original%20Firmware/ReadMe.txt

This is a deal breaker for me. The extra channels aren't worth the simulators being hard coded. I'm trying to roll back to stock firmware.