benb0jangles/FlySky-i6-Mod- #69: No option to select a COM port above COM 8

aakashpolra, 2017-04-11:
My USB to Serial adapter is assigned as COM 10 on Windows. Unfortunately the exe only has options up to COM8. How about having a free-form text field to enable connecting to arbitrary COM ports?

MathijsG, 2017-04-11:
Not really a solution program wise, but if you go to device manager and go to the properties of the port -> port settings -> advanced you can choose a different com port for it.

qba667, 2017-04-11:
@aakashpolra have you tried https://github.com/ThomHPL/BaseI6_updater and FW from rcgroups?
benb0jangles is updating the GUI updater just time to time.

Gryd3, 2017-04-11:
I have two other solutions for you as well.
The 'Native Linux App' is a cross platform app, and the author has pre-compiled bins for 64-bit Linux and Windows here : https://github.com/mhils/flysky-updater/releases
It's a command line utility, but you can choose the COM port you wish like qba's suggestion.

Additionally, you can go into the Windows Device Manager and manually re-assign a COM number less than 8 to your serial device for use with benb0jangles' GUI.

benb0jangles, 2017-04-11:
"@aakashpolra have you tried https://github.com/ThomHPL/BaseI6_updater and FW from rcgroups?
benb0jangles is updating the GUI updater just time to time."

-I'm actually in agreement with using this updater for future firmware because telemetry screen will need to be configured as user wants it before uploading. This updater will make this easier to accomplish.

aakashpolra, 2017-04-14:
Thank you guys for the solutions. I have managed to update it and looking forward to try it out. Appreciate your replies and hope this thread will help someone stuck in the same situation.