benb0jangles/FlySky-i6-Mod- #74: tx battery sound + bind bug

zelwake, 2017-05-11:
Can I disable or adjust when will tx beep? Right now when tx battery drops by 0.01V there is a beep which is higly annoying. Might be fixed in newest qba667 version, havent tried yet.

Binding. That thing is pretty bad atm. Had to turn on in second menu, hit Bind, held bind button on rx, plugged battery, tx showed "binding ok" but couldn't arm. Had to turn off and on tx and hit Bind again (no turning on/off quad) and it finally took signal. Was scratching my head for 10 minutes before I figured this out.

qba667, 2017-05-18:
@zelwake I am trying to figure out what is causing this issue. Can you give me a hint? What was last working version?

zelwake, 2017-05-18:
this is the first version (here) of yours project i use, before I was using 10ch timer...actually it is working, kinda harder to bind (more steps), but its working
wonderful, setting tx alarm got rid of the annoying reminder beep, thanks for that tip

qba667, 2017-05-18:
@zelwake 10ch timer was also mine - so it was working before. I have to verify all changes.

zelwake, 2017-05-18:
@qba667 really? thats nice!....anyway, managed to test ia6b receiver and didnt need to rebind, it can also be just x6b receiver issue

qba667, 2017-05-22:
@zelwake 
Binding issue resolved can you try:
https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch%21/page128

zelwake, 2017-05-25:
@qba667 
alright, bind button works nicely now, rebinded (rebound?) both ia6b and x6b at first try, gonna try later with afhds receiver on my micro

zelwake, 2017-05-25:
okay, working with AFHDS too, I think problems are solved

closed 2017-05-25 by zelwake