benb0jangles/FlySky-i6-Mod- #75: all  this firmaware for i6 is useless

Juli79, 2017-05-20:
for me and others which  have a naza flight controller ...is useless without the endpoints for all channels

Gryd3, 2017-05-20:
Juli79, then the solution for you, and others like you is pretty obvious.
Don't use it. The mod works amazingly well for flight controllers that support serial TX, and I can certainly claim that the function I gain with the mod is well worth the effort and mod when used with a serial TX capable FC and firmware (Naze32 and Skyline32 personally).

qba667, 2017-05-22:
@Juli79 there is already source code published to handle channels 7-10. Just waiting for some willing to invest time and rework the method. EEPROM access is also provided. Instead of complaining start working and provide something to the community.

closed 2017-05-23 by Juli79

alandroid, 2017-12-24:
@Juli79 "Useless for Naza". For everyone else, much useful, this software is #yoda