benb0jangles/FlySky-i6-Mod- #76: voltage alarm

rcstuff, 2017-05-24:
hi great firmware. im using 10ch timer mod and i have an idea how to improve it. i run diferent bateries 3s and 4s, on my transmiter i have diferent model profiles for same rx , one for 3s and one for 4s. problem is whene i forget to swich it :) its ok whene my tx is set on 4s voltage alarm buzzes if i plug 3s but other way around is diferent if my tx is set to 3s voltage i get no alarm whene i plug 4s battery. so my idea is to buzz alarm whene voltage is higher than set value.

qba667, 2017-06-12:
@rcstuff  sound good, added to my to to list.

LazaroFilm, 2017-06-29:
Definitely +1 on that. I had to failsafe my 4s when I realized that I was killing it, waiting for the alarm to go off. I guess there is no way to make it work like Betaflight that figures out the cell count from the current voltage right?