benb0jangles/FlySky-i6-Mod- #79: Tx Voltage adjust - is there a way to acess that menu

yo8aiv, 2017-06-23:
Tx Voltage adjust - is there a way to acess that menu. I run different baterry types and sometimes need recalibrating for NiMh or LiPo, or maibe there is a way of selecting betwee baterry types.

See video for better explanation.
https://youtu.be/o5LPqV-TWLQ

Is there a way i can acess the **Voltage adjust for the transmitter** on FlySky I6 / Eachine i6 ?

Thanks in advance.
Cheers.

qba667, 2017-07-07:
You want to have 2 different batteries? I am switching between 1s and 2s packs as well, is the calibration needed or just alarm value?
Create issue in my repository because here I can not handle it correctly.
https://github.com/qba667/FlySkyI6

yo8aiv, 2017-07-10:
yes. i use NIMH and 1s or 2s. Alarms are a must since we swap batteries alot.
 just upgraded to that firmware and it's great.
Thanks. Fantastic job.