benb0jangles/FlySky-i6-Mod- #83: a8s

miko671995, 2017-07-25:
Hello will it work with a8s receiver ?

qba667, 2017-07-28:
@miko671995 it should be working fine.

miko671995, 2017-07-29:
> ** Fri, 28 Jul 2017 11:55:52 +0000 (UTC) - "benb0jangles/FlySky-i6-Mod-" <flysky-i6-mod-@noreply.github.com>, "benb0jangles/FlySky-i6-Mod-" <reply@reply.github.com> **
>
> @miko671995 it should be working fine.

the telemetry is possible too ?

lrusnac, 2017-08-01:
@miko671995 the a8s has no telemetry, it's just 1-way comunication

qba667, 2017-08-01:
@lrusnac correct but this is only limitation of implementation in RX, I was able to made IA6B single wire for telemetry and channel data. But in case of other RX there is much more work because telemetry implementation is missing.

lrusnac, 2017-08-01:
@qba667 are you saying the hardware of the a8s allows to communicate back with the tx but it's just a firmware on the a8s that doesn't allow that?
As far as I see it, the a8s does not have a tx module to send back anything, unlike the full sized modules like IA6B.

qba667, 2017-08-01:
@lrusnac  I was experimenting with A7105 modules they are bidirectional (I have used them to trigger Yongnuo flashes) goebish is also using such module for 2 way communication in Deviation TX.

lrusnac, 2017-08-01:
@qba667 that is brilliant, I need to look into it, having some kind of data back from these tiny receivers will make them really awesome

qba667, 2017-08-01:
There is even one nice implementation with telemetry - 
https://github.com/vladisenko/FlySky-AFHDS-A2-Receiver

Of course to get telemetry from outside you need to change serial port behavior - what MCU has a8s? Does it support bidirectional UART? If yes other pin must be used if not 4k7 resistor between serial RX and serial TX will just work. I don't have such RX but for 90% they used a7105/6. Can you provide some pictures of the RX?

lrusnac, 2017-08-01:
I also found this https://github.com/cassm/a7105 that would potentially allow to create an open source receiver.

I will post some pictures and MCU details later today when I get home

lrusnac, 2017-08-01:
@qba667 So, I have both: A8S and the FS-RX2A (I had the A8S wired so a bit difficult to take pictures https://imgur.com/gallery/7Jo8c )

the chips look the same, they both use A1706 for the RF and stm32f031k4 MCU

qba667, 2017-08-02:
@lrusnac so open FW is possible, only small subset of packets IDs is already known but it seems to be enough to implement basic functionality.

lrusnac, 2017-08-03:
@qba667  I should try it but at the moment I have no st-link device, should get an f4 discovery

qba667, 2017-08-08:
@lrusnac discovery will be fine - I have spare development boards if you want they have ST-Link build in.

lrusnac, 2017-08-08:
@qba667 that will be really cool