benb0jangles/FlySky-i6-Mod- #89: Eachine E010S Binding Issue

azz1266, 2017-08-26:
I'm trying to Bind my Eachine E010S micro quad and with the modded Flysky I6 firmware, it will not bind (AFHDS mode), if i revert the firmware it will bind. Can you please help / advise.

theoldfart, 2017-08-26:
I had the same problem.  set switches and gimbles to the OK position, ie where you have to put them after switching on. THEN switch on whilst holding bind button.

azz1266, 2017-08-26:
what do you mean by the OK position? i've tried binding with the switches in the "off" position in order for the controller to turn on with no luck.

azz1266, 2017-08-26:
all sorted now, i had to go into the hidden menu, select bind (after changing the current model to AFHDS), the blue L.E.D shows it has binded, then you have to power cycle both drone and transmitter, change model then change back and it fires into life!, bit of a fudge around but it seems to work, if next time you reconnect and its not working, switch model then switch back and it should fire up fine again, would be good to have it bind like a normal receiver though :) thanks for the modded firmware by the way! saved me a few quid in upgrading!

qba667, 2017-09-18:
@azz1266 what FW version have you used?

azz1266, 2017-09-18:
The latest one available with the timer mod etc

qba667, 2017-09-18:
@azz1266 Have you used console based updater from my repo?