benb0jangles/FlySky-i6-Mod- #91: FLY-SKY I6 0.00

R3dSmoke, 2017-08-29:
Have tried to update my Tx many times but every time it gets to 100% the version changes from 1.10 to 0.00 and bricks my Tx. Any ideas?

qba667, 2017-09-18:
@R3dSmoke please provide more details about hardware used (also usb to serial dongle).

agteigarnet, 2017-10-30:
Have experienced exactly the same. No problem flashing the original when the TX is bricked.
Tried v1.5  very many times, v1.4 few times always similar.
Most of the time not getting to 100% still nothing works(bricked).
When bricked, simply flash the original firmware no problem at all, always 100% success and fast.
Windows 7, Arduino uno as dongle.

Disjointsnow, 2017-12-04:
Hi,

Is this issue resolved? 
I got the same problem after program but I did not get a chinese massage before programming. It just start and go to 100% and then nothing happends, now my display of my fs i6 wont go on

qba667, 2017-12-04:
@Disjointsnow, @agteigarnet sorry for late answer - they only idea coming to my mind is to use different updater, have you ever tried console updater from https://github.com/qba667/FlySkyI6?
@Disjointsnow  you can still recover the unit in silent update mode.

Disjointsnow, 2017-12-18:
Hi, 

I do need to edit one more thing before i did the upgrade, the jumper was on 5v instead of 3.3v.
Does this change the situation?
I haven't tried anything yet because its all new for me.

How does the silent update mode go on? Or is that with the trim switches and then turn it on?

Here are some pictures of the dongle I used and the number of the transmitter.
[image]
[image]
[image]

Or is the transmitter now finished because I used the jumper on 5v?

agteigarnet, 2017-12-18:
Yesterday I tried the NO_SWE_14_channels_flysky-updater_x64  updater and it worked fine.
Used Arduino for serial/usb interfacing.
Now working on using the extended channels :-)

qba667, 2017-12-18:
@Disjointsnow it is possible that you burned UART in MCU - but it is still wort trying - use console updater.
Once I have connected something around 8V but only for few seconds - the unit is still working.
@agteigarnet there will be one update more allowing endpoint and subtrim configuration.

szafranski, 2019-02-01:
For all people with this issue - I was fighting with this all day. The solution that worked for me - and you have to know that I tried almost every other metod that is allready aviable online - was to plug FlySky to my friend's computer with Windows 8.1. I was trying several programs, hex files, versions, drivers. Solution was just to plug my FTDI adapter do another machine with different OS (mine is working on Windows 10). Drivers were automatically downloaded by Windows after plugging. Then I connected radio in "firmware mode" (trims and power), opened same program I was using that earilier bricked my radio (10ch mod v1.5 etc). Clicked "open port", FlySky 0.0 showed up, thent clicked "Program" and voilla! My radio is back, fully functional with additional aux channels. 

As far as I know those 10ch mods are designed for beeing flashed on stock firmware (1.1 or sth). So when you try to flash this mod on allready updated version it can cause some issues. 

Give it a try, maybe it would work on virtual machine too. Cheers!

Benioup, 2020-07-13:
> @Disjointsnow it is possible that you burned UART in MCU - but it is still wort trying - use console updater.
> Once I have connected something around 8V but only for few seconds - the unit is still working.
> @agteigarnet there will be one update more allowing endpoint and subtrim configuration.

Hey, i think i may have burnt my UART in the MCU since i am using a ftdi adapter. Do you know how to fix this? I'm asking about it because i am trying to update my fsi6 firmware for several weeks now but i still can't do it... anyway thanks

szafranski, 2020-07-13:
Sounds fixable

Benioup, 2020-07-19:
@szafranski is it really fixable? if yes how do i do it? thanks for helping

szafranski, 2020-07-19:
Could you explain what is your issue further?

Benioup, 2020-07-19:
@szafranski I have a flysky fsi6. I am trying to do a firmware update via FTDI adapter. I followed every step with care. On the programmer on my comuter, i can open the port but it won't show version number. As i saw in the discussion above, i think i may have burnt my UART for the trainer because i had the ftdi bridged on 5V instead of 3.3V this whole time . Do you know how i could fix it? or will i never be able to update my fsi6?

szafranski, 2020-07-19:
I would try on another PC. I know it sounds stupid but those drivers on
Windows... Man... :)

Benioup, 2020-07-19:
i have an old pc a windows 7 you think this is alright?

szafranski, 2020-07-19:
Hard to tell. Random issues with drivers are unpredictable.

XeroDays, 2020-12-16:
This issue is still not resolved, I am using windows 10, and facing the exact same Error.
**The CLI firmware works but i am unable to upload the original firmware to transmitter now!**

taurusturk, 2021-03-07:
I wanted to upgrade from 6 channels to 10 channels. FLYSKY i6 remote does not work. does not upgrade. it didn't happen with the data cable. did not work with ardiuno, can anyone help: ((((

XeroDays, 2021-03-07:
@taurusturk  can you please give some detail about what is happening while you install?  I did have the same issue and solved it then.

taurusturk, 2021-03-07:
@XeroDays thanks for the answer. I wanted 10 channel upgrades from 6 channels. Installation was incomplete. the remote is turned off. does not open. I could not open it in any way. I've tried almost everything: (((
[image]

XeroDays, 2021-03-08:
@taurusturk   I made this small txt note for installing firmware in future. Take a Look.

`The FlyPlus  firmware works fine with 14 channels..
> Other softwares give issues
- Make sure the Wi-Fi is not (set Metered) i don't know what's this issue

Flysky Secret Menu if working fine.
- Left Stick (Bottom left)
- Right Stick (Bottom left)

Flysky if Breaks ( When you Power On transmitter and shows nothing)
- Trimmer (YAW Right)
- Trimmer (Throttle Down)
and Power ON
Note : Screen in flysky remote will be off but it will be in "UpdateMode", don't worry.

how to do it?
- insert the ftdi port in the computer,
- open the software ( cli console)  (Flysky Plus) Note: In my case, other don't work somehow.
- select any of the .bin file.
- select port
- and it will automatically initiate and install.

if error:
-if the cli crashes
- remove and re-enter the USB of ftdi port`

Download Flysky Plus files:
[http://skysoltech.com/data/flyplus.zip](http://skysoltech.com/data/flyplus.zip)

taurusturk, 2021-03-09:
@XeroDays Thank you for your interest.  
[image]
yes finally uploaded. my data cable is faulty. the problem is solved with the new cp2102 module (usbtoTLL). 14 channels have been upgraded.