benb0jangles/FlySky-i6-Mod- #92: No bind problem hali9 10ch Update

BASSBEAT4LIFE, 2017-08-31:
First THANKS to the creator of this mod it is great BUUUUUUTTTT i cannt bind my quad with it -.- so all the good stuff is sensless for me -.-

I switched in to AFHDS prot cause this is what my flight controler use.... I did the binding like i always do... set the radio on bind, set the drone on bind, switch of the radio, than the drone and after restarting the controller its bind... but after this mod nothing happens.... i flashed back to a older 10ch mod (but less funktions....) and there i get the bind really quickly.... i also flashed back to stock and tried there an also with that software the binding wos easy but i'm not able to bind with this mod -.- I use a Furibee F3 32bit board

Please please help me i whant to use THIS mod !!!!

azz1266, 2017-08-31:
Have a look at my e010s thread i found a fix for mine tiny whoop binding issue

BASSBEAT4LIFE, 2017-08-31:
i did not realy understood how it works... (sry for my bad english) ... how do i get in to the "hidden menu"? 

all sorted now, i had to go into the hidden menu, select bind (after changing the current model to AFHDS), the blue L.E.D shows it has binded, then you have to power cycle both drone and transmitter, change model then change back and it fires into life!, bit of a fudge around but it seems to work, if next time you reconnect and its not working, switch model then switch back and it should fire up fine again, would be good to have it bind like a normal receiver though :) thanks for the modded firmware by the way! saved me a few quid in upgrading!

azz1266, 2017-08-31:
Power off transmitter hold both sticks to the bottom left and power on go down to bind

BASSBEAT4LIFE, 2017-08-31:
ok thanks it worked.... BUT how can i avvoid the annoing switch between models ? because every time i turn off and on my receiver i have to do that to get it bind again...

azz1266, 2017-08-31:
I have to do that at the moment not a great way of doing it but it works! I have requested this to be looked at in the next update. Glad your up and running!

BASSBEAT4LIFE, 2017-08-31:
other question, how do i start the timer /alarm under the AFHDS prot? i dont get it...

azz1266, 2017-08-31:
If you look at the forum I have raised this as a bug as it currently does not work in afhds.

BASSBEAT4LIFE, 2017-08-31:
did we know when we will get a new update ? And if the one that writes these softwares is reading thank you for yor work !

azz1266, 2017-08-31:
Unsure you will have to ask the contributors agreed thank you for the contribution!

qba667, 2017-09-18:
@BASSBEAT4LIFE @azz1266 this will be minor change - I will try to add it soon.