benb0jangles/FlySky-i6-Mod- #95: fsia6 binding

darihr2, 2017-09-16:
Hi...i have problem with this reciver...i connected it in pmw mode and when pover is added red led on rx blinks like it is in binde mode...i put radio to bind mode and it connects but when i unplug everythyng i plug it again reciver is stil in binde mode(fast blinking red) can someone help me...thanks

qba667, 2017-09-18:
@darihr2 please more details - FW version, RX type. Have you removed the bind jumper after binding?

darihr2, 2017-09-18:
Hi...10channel mod with fsia6 is not working i have two same recivers and they have same simptoms
But my fsia6b works great

darihr2, 2017-09-18:
No my is 10ch heli something not 14ch

qba667, 2017-10-04:
@darihr2  then use updated version it will work.