benb0jangles/FlySky-i6-Mod- #98: Will this firmware work with ia6 receiver

batata004, 2017-11-13:
Hi,

I really dont need the 10ch of this firmware -> https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Timer%20Mod%20i6%20Updater/10ch_Timer_MOD_i6_Programmer_V1_4

I just need the other cool things, like the battery alarm over 12v and small other things.

My question is this: will the FS-IA6 receiver stop working if I update my transmitter to the firmware above? I mean, not the IA6B, I am asking about the IA6 -> https://produto.mercadolivre.com.br/MLB-750649444-receptor-flysky-fs-ia6-com-dupla-antena-_JM

Also, is the V_1_4 the latest version I should install in my transmitter?

qba667, 2017-11-13:
@batata004 IA6 will work fine - in version you mentioned there is possibly bug related to AFHDS receivers. 
But use latest version from my repository.

batata004, 2017-11-13:
@qba667  I will install your version right now! I will just wait an answer from you on my other thread. As soon as I install it I come here to tell you if evertying went fine :)

closed 2017-11-13 by batata004