benb0jangles/FlySky-i6-Mod- #101: Bug getting Err%

batata004, 2017-11-22:
Hi,

I discoreved a weird behaviour (that I classify as a bug): after binding transmitter/receiver turn off everything. Turn on the transmitter and after 5 seconds turn on the receiver. You will see that no values for Err% or intV will be displayed. With the stock firmware of the TGY-I6 radio this does not happen, it only happens with the latest firmware update. I am using the ia6B receiver.

With the stock firmware, even if I turn on first the radio and then the receiver, there is no problem. This is not a big deal, I can turn on first the receiver and only then the transmitter but I think this could be the effect of a more serious problem that I reported in another thread.

qba667, 2017-11-23:
@batata004 please provide version information.

batata004, 2017-11-23:
@qba667 I am using the latest of the latest, the one you told me in another thread to use. I am using your firmware, not the stock nor the others around!

qba667, 2017-11-23:
@batata004 okey got it, one more question are the values from I6 send to receiver correctly?
Or there is no communication at all? I will check the issue at evening.

batata004, 2017-11-23:
@qba667 Thanks! If you simulate the order of procedures I said you will see that the intV and Err appear in the home screen BUT they have no values set (not even zero). It just shows:

intV
Err

Nothing more. In the sensor list I chose only intV and Err to display in the screen ok?

qba667, 2017-11-23:
@batata004 got it - I will try to find the reason.

batata004, 2017-11-23:
:) thank man!

qba667, 2017-12-04:
@batata004 I was able to reproduce the issue. But the reason is not known yet.

batata004, 2017-12-04:
@qba667 thanks man, I am really glad you reproduced this cause I thought my transmitter was faulty! I am sorry that the reason is unknown but at least now we know how to deal with it. I think it's valid mentioning that using the stock firmware this problem does not happen!

One more thing my friend: I setted battery alarms and they work great but there is some weird problem. Suppose I configure my low medium alarm to 3 and 3.1 v.

If my voltage drops too fast from 3.1 to 2.9 the alarm will not beep! I think the alarm should not check if it is between the low and the medium. It should just chekc this: is the voltage below the alarm? Yes? Play the alarm.

You can easily check this bug with a much easier method: set your low alarm to 3.9v (for example) BUT connect a battery that is already below 3.9v (like 3.8v or below). You will see the alarm will not play! It should, right?

1074748905, 2020-04-09:
@qba667 i have the same bug.  FS- i6 TX(your 1.75 firmware) with  ia6B RX

AND when home screen display intv ，rssi err SNR but no values，，i find there is a great latency  to control   servos，i can easily  find servos moving NOT smoothly。
AND after few seconds or minutes ，intv ，rssi err SNR  values  will appear.