benb0jangles/FlySky-i6-Mod- #102: Firmware Version Number won't show + "Device is not connected"

slayervictor, 2017-12-02:
Hello.
I am trying to get the latest firmware update on my flysky fs-i6, but when i connect to a com port, for me thats com 1, it won't show what version of firmware im using. When i click "program" it just says "Device is not connected" although i have connected it. What do i need to do?
yes i have remembered to run as a admin.

Heres a photo
[image]

slayervictor, 2017-12-02:
[image]

qba667, 2017-12-04:
@slayervictor  ensure that device is listed in device manager - check com port number. Or even better use console updater from here https://github.com/qba667/FlySkyI6

ivanPdmnt, 2017-12-13:
@slayervictor did you manage to find the solution?? Im in the same situation... Port opens but doesnt recognise the device. thanks!

qba667, 2017-12-15:
@ivanPdmnt, use console updater, select correct COM port, check connection, monitor serial interface traffic, provide logs:)

slayervictor, 2017-12-15:
@IvanPdmnt No i havent found a solution, but maybe if i figure out how to use the console updater i can show u
@qba667 can you explain how to use the console updater?

slayervictor, 2017-12-15:
@qba667 and which version i will need to use on the fs-i6

qba667, 2017-12-18:
@slayervictor last one - 1.7.1 or 1.7

amckee, 2018-07-12:
I ran into this as well. I swapped the tx and rx wires and it was able to successfully update the firmware.

Benioup, 2020-06-24:
@slayervictor did you find any solution. I know im late but any solution...? that would be great thanks alot!! :)

ivanPdmnt, 2020-06-24:
Hello!
yes I managed to solve the problem, but honestly I dont remember how I did
it, too many months since then... i sold the radio one year ago, but it was
working fine with all 10 channels. Thanks anyway!

Benioup, 2020-06-25:
@ivanPdmnt too bad :/ if you remember with any luck please contact me that would awsome!! thx :)

EvgeniyJarsky, 2021-01-30:
I have the same problem, but solved the problem. It was bad contact. You mast try any times to connect flysky to computer. I did it 8 times and a did it at last time!!!

taurusturk, 2021-03-07:
I wanted to upgrade from 6 channels to 10 channels. FLYSKY i6 remote does not work. does not upgrade. it didn't happen with the data cable. did not work with ardiuno, can anyone help: ((((

mahmud-hasans, 2021-03-23:
@Benioup I got it working by watching a comment from RCgroups. I connected the Transmitter without turning it on. Then I attached all the cables. Only after attaching all the cables I turned on the Transmitter in Update mode. Then it worked fine.