benb0jangles/FlySky-i6-Mod- #106: Any plan for fs-i6x

elbowz, 2017-12-27:
I would really like have the alarm and timer on my FS-i6x, your mod is so great!

Any dev thinking to do the same work on i6x?

thanks for your hard works!

elbowz, 2018-01-03:
Thanks for answer!
I hope someone other...never say never ;)

I don't have the skill for do this works, but I guess some (many) others guys wait for this.

Happy good year!

qba667, 2018-01-19:
@elbowz 
I have changed my mind.
Please see:
https://www.rcgroups.com/forums/showthread.php?3020949-FlyPlus-firmware-for-FlySky-I6X

Sabbadin, 2018-06-03:
@qba667 

Hello!

I bought an FS-i6X and frankly, although it's simple, I found enough resources for a radio of its size. I'm really missing a flight timer.
Is there any way to update this in it? Any new firmware I'm not finding? Would you help me? Thanks in advance!

Sabbadin, 2018-06-03:
@elbowz 

Hi! Did I find any updates with TIMER for the F-i6X? Thank you!

elbowz, 2018-06-03:
For what I have understood qba667 have released a modded firmware with rssi on channel 11 plus alarm.

Now he have preferd to work on OpenTx port.

Find more update on patreon or rcgroup topic.

closed 2018-06-03 by elbowz

Charli-code-code, 2025-09-02:
We are in 2025... Anyone have the solution to have the timer in FLYSKY FS I6X? I dont want to change the firmware to Opentx