benb0jangles/FlySky-i6-Mod- #107: head tracking via trainer port

unitware, 2018-01-07:
Hi, I've built a servo controlled camera gimbal for the FPV cam that I'd like to control with the head tracking sensors that are built into my FPV googles (fatshark teleporter v5). 

The googles output a PPM signal that can be input in the trainer port with PAN/TILT on channel 5/6. The Flysky does not handle this well, I believe that activating trainer mode just puts channel 1-4 on the normal channels.

A mod that mixes the PPM5/PPM6 to channel 5/6 would be great, do you think that it is doable?
Or maybe mixes the PPM5/6 with VRA/B and then can be mixed with any channel?

unitware, 2018-01-07:
@qba667 I just saw your repo again, does it support this already? And if so, does it work without updating the receiver?

qba667, 2018-01-07:
@unitware yes PPM input is supposed. The radio can handle 6 channels of PPM input. As channel value PPM6, PPM5 and PPM4.

unitware, 2018-01-08:
Great, is it possible to just reflash the transmitter or do I have to upgrade all of my receivers?

qba667, 2018-01-08:
@unitware just the transmitter - they are 2 versions of FW for standard units and units with additional switch SWE - ensure that you use correct one. Please verify that your goggles are providing PPM signal recognizable by the transmitter - and let me know - small changes related to time specification of the PPM input are possible.

unitware, 2018-01-08:
excellent, this one then? https://github.com/qba667/FlySkyI6/releases/download/1.7.2/1.7.2.zip, and the bin file without "-swe"

qba667, 2018-01-08:
@unitware you can try this one - it is last build but so far seems to be stable.

closed 2018-01-08 by unitware

qba667, 2018-01-08:
@qba667 in case of any issues just let me know.

unitware, 2018-01-08:
absolutely, thanks so far!

unitware, 2018-01-08:
is there any linux updater by any chance? or can it run via wine?

qba667, 2018-01-08:
@unitware try this one https://github.com/mhils/flysky-updater

destarita, 2018-02-17:
@unitware how did you connect your the headtracker sensor to the FS-I6?

qba667, 2018-02-17:
@destarita using PPM input

unitware, 2018-02-18:
Yes, I used PPM.
Took a 2m s-video cable and rewired one end.

Pinouts of my fatshark googles:
https://4.bp.blogspot.com/-2ZeB0dyndpg/WKT10SpypvI/AAAAAAAAeDk/Ch5z7w0fVE8Cnc2x_5zly2XH4jl4AQ0YACLcB/s1600/datacablespec_A_zpsf1c7ad15.jpg

Pinout of the flysky connector:
https://lh3.googleusercontent.com/-wH6cknPGBos/V5zCTMPpRhI/AAAAAAAADTY/4HavIuulRUo9pxkKar0cQdmetnyy39fIwCCo/s0/Trainer%2Bport%2Bpinout%2B2.jpg

GND is on the shield, you need that and the ppm pin.

destarita, 2018-02-20:
@qba667, @unitware thanks a lot. It worked perfectly! I connected my DIY headtracker to my TGY-I6... The only thing is the timer,  is no working anymore after the FW update. Any suggestions?

Kind regards!

unitware, 2018-02-20:
Glad it worked!

No clue of the timers, check the issue list and raise a new one if needed. Don’t forget to specify the version you are using. 

Skickat från min iPhone

duwapp44, 2019-09-30:
Hello everyone!
I have been trying to make an arduino DIY headtracker work on my flysky i6. I have very little understanding on how trainer mode works. I have downloaded and installed the firmwware posted previously but I do not know how to assign those ppm signals to channels 5 and 6 of the transmitter, while mantaining channels 1 to 4 free to control the plane (when activating trainer mode those four first channels seem to become "locked"). Could someone tell me the process of configuring the transmitter? That would be much appreciated.

unitware, 2019-09-30:
Trainer mode is something different, turn that off and you should be fine. 

Skickat från min iPhone

Nir152, 2020-07-22:
qba667  & unitware ;
Can I conect the Fatshark trinity head tracker which is outputting or standard cycle (8CH PPM, 20ms)
 rapid cycle (3CH PPM, 8ms)  to the input ppm pad on the mainboard and output the **total ppm(controls and HT) by the ppm output pad** to my Frsky X9D remote via trainer cable?