benb0jangles/FlySky-i6-Mod- #118: FS I6 SBUS OUTPUT  ON TX ?

mikerc2017, 2018-08-22:
HELLO DEVELOPERS 
FS I6 SBUS OUTPUT  ON TX  INSTEAD PPM ONLY ?

winiak, 2018-09-14:
I would also appreciate having RC channels send to external TX module by UART (ibus, sbus, etc..) no just by PPM. I'm working on LoRa communication using PPM as input but having iBus would improve the latency.

qba667, 2018-09-17:
@mikerc2017 
@winiak 
This should be easy hack to existing firmware -  create ticket in my repository - maybe during coming winter I can implement it.

mikerc2017, 2018-10-25:
@qba667 

do you work with it ?  Sbus or CROSSFIRE protocol for tranier port .   
this guy fly with Modul Longrange mod PPM 

https://www.youtube.com/watch?v=m4Gc57gW3oQ&fbclid=IwAR0yW0ppvTZuW7z4ht84AGxk_07x3hvSFy7j6GxxVTYV7Lc393g5xDwZgHo

widi22010115210018, 2018-10-25:
@mikerc2017 I don't understand what are you trying to say, but in the video he just use ppm, basically plug and play, he just add Connector, it basically plug and play

mikerc2017, 2018-10-26:
> @mikerc2017 I don't understand what are you trying to say, but in the video he just use ppm, basically plug and play, he just add Connector

i mean use SBUS or crossfire instead PPM as video .

aleksirastas, 2019-06-17:
> > @mikerc2017 I don't understand what are you trying to say, but in the video he just use ppm, basically plug and play, he just add Connector
> 
> i mean use SBUS or crossfire instead PPM as video .

Did you find any solution for this? 
I was thinking that could it be possible to use oscilloscope to find out where there is 100k baud rate signal?

winiak, 2019-06-18:
you won't find the IBUS just like that. To have IBUS or SBUS you need to use UART port (TX-RX) and implement one of the protocols to work. It's easy for one direction - without telemetry. With telemetry there is much more to do....

aleksirastas, 2019-06-18:
Do you mean that there is no sbus/ibus signal transferred on the board? I think that sbus is only signal using 100k baud rate so it would be easy to just look for a signal that would stay on scope when using this rate.  
I know nothing about how to mod i6 tx/rx configuration. Telemetry is not mandatory because im not using it currrently. 

Aleksi

From: winiak
Sent: Tuesday, 18 June 2019 18.43
To: benb0jangles/FlySky-i6-Mod-
Cc: aleksirastas; Comment
Subject: Re: [benb0jangles/FlySky-i6-Mod-] FS I6 SBUS OUTPUT ON TX ? (#118)

you won't find the IBUS just like that. To have IBUS or SBUS you need to use UART port (TX-RX) and implement one of the protocols to work. It's easy for one direction - without telemetry. With telemetry there is much more to do....