benb0jangles/FlySky-i6-Mod- #120: IRX6 "rudder quads" stop working

konstantint, 2018-10-01:
IRX6 is a multi-protocol module, which plugs into the trainer port, reads the PPM signal, and lets you control various toy quads. In order to select the protocol one needs to provide an appropriate gesture when powering up the module (e.g. "aileron up" would enable one protocol, "rudder right + aileron left@ another one, etc).

It seems that with the new firmware the IRX6 stops recognizing any protocols which have the rudder involved in their switch-on gesture.

Does anyone know the reason for this behaviour and a possible way of fixing it?

konstantint, 2018-10-01:
Interestingly, installing the original firmware (from this repository) does not seem to fix the problem any more.

qba667, 2018-10-04:
@konstantint, can you verify value of rudder stick?

greekuser, 2020-07-02:
Hi, has this issued been fixed with recent firmwares? I wanna know before flashing. Thanks