benb0jangles/FlySky-i6-Mod- #122: Bricked FS-I6?

DanBoghean, 2019-05-17:
I tried to install the firmware via a mac using these instructions: https://redirect.viglink.com/?format=go&jsonp=vglnk_155806490735711&key=5e5e6e04e939b9674c384ca8b03117f4&libId=jvrjm5ol010248ul000DAs3snfefm790l&loc=https%3A%2F%2Fwww.rcgroups.com%2Fforums%2Fshowthread.php%3F2486545-FlySky-FS-i6-8-channels-firmware-patch%2521%2Fpage243&v=1&opt=true&out=https%3A%2F%2Fwww.jaylagare.com%2F2018%2F10%2Fflashing-flyplus-with-a-macbook%2F&title=FlySky%20FS-i6%208%20channels%20firmware%20patch!%20-%20Page%20243%20-%20RC%20Groups&txt=https%3A%2F%2Fwww.jaylagare.com%2F2018%2F10%2Ffl...ith-a-macbook%2F

The first time it got to 100%, but it was stuck on the throttle warning when turning it on. So I tried to redo it using the Trim trick to enter firmware update mode. 

This time, I chose the SWE version. It got to 88% and gave me this error: "Error flashing firmware: EOF"Now I can't turn it on and even using the trim/yaw trick isn't working. I get "Cannot ping the remote: EOF."I connected it to a Windows 10 computer and tried running the .exe, but it says the device is not connected.Any ideas? Or is it dead?

qba667, 2019-09-13:
@DanBoghean it is not, but probably only J-link, ST-link can restore it.

masonhorder, 2021-01-20:
any updates, my remote works as normal just wont recieve the rx data. tried switching cables and everything but no luck.

error: 

Cannot ping the remote: EOF. Is the remote in firmware update mode?